jgarff/rpi_ws281x #142: ImportError: No module named neopixel

eurasianjay, 2016-12-13:
Hi, ...
I'm trying to light up my WS2812B LED stripe with my RPi 3.
I have already followed the instructions on https://learn.adafruit.com/neopixels-on ... i/software
but whenever I try to run the "strandtest.py" I get the error message:

File "strandtest.py", line 8 <module>
from neopixel import*

ImportError: No module named neopixel

I have already downloaded and installed git clone https://github.com/jgarff/rpi_ws281x.git step by step... 

Do you have any idea, why I still get this error message?

Thanks a lot in advance !
Cheers, Jay

freakyuser-osm, 2017-02-05:
You have to install the python dev packages:

`sudo apt-get install build-essential python-dev git scons swig`

Then run scons etc again.

See as well adafruit:

https://learn.adafruit.com/neopixels-on-raspberry-pi/software

jandolina, 2017-09-29:
Is it possible to install the library globally or move it work somewhere other than the examples folder?

marbaque, 2017-11-07:
eurasianjay you need to run the strandtest.py example directly from the terminal

pietrodn, 2018-01-02:
The instructions about how to install the library are now in the python/README.md file.
I think this issue may be closed.

Gadgetoid, 2018-03-06:
Agreed!

closed 2018-03-06 by Gadgetoid

PanderMusubi, 2018-05-10:
The root cause is usually a kernel upgrade. In that case, only the Python install command needs to be rerun. The safest is to firstly also rebuild all.