jgarff/rpi_ws281x #143: Throws exception at begin() when used as part of a Flask web server.

jeristotle, 2016-12-18:
I know this goes beyond this module, but I know this functionality is pretty desirable.

I'm trying to get the LEDs to operate in response to calls to a web server.  I set up a route (just "/") in Flask, then go into that route's associated method def, in which I call up a strip and do something to it based on that particular route.  For starters, I just wanted to turn a pixel on.

NOTE: I've used the Adafruit library on Arduino and Raspberry Pi before, so I'm used to the code.

When I instantiate a strip (code copy and pasted from the python library... thank you very much, btw...), I get an exception that I traced back to the neopixel.py file, begin() and the like.

I did some reading because I tested the code out, and the exact same code works from the command line but not through the web server, and it's all about permissions... (I think).  Messing with the GPIO pins is not something they'll let just anyone do, so when something without adequate permission tries to access them, it fails.

So I tried using sudo on the command line when I started the web server.  No dice, presumably because the library files didn't have the appropriate permissions.  So I changed neopixel.py and rpi_ws281x.i (root:root, 4755), still no joy.  Kinda shooting  at anything, figured I'd try the experts.

I'm relatively new to Raspberry Pi and real Python programming despite my specific use of the libraries to make things go all glowy and sparkly, so please be gentle with me.

Thanks for any help you can give!

Have a great one.

linguistrix, 2017-04-09:
Can you post your Python flask code here?

Gadgetoid, 2018-03-06:
I'm a bit late to the party here, but if you're running Flask in debug mode with Reloader, or otherwise running it with Reloader enabled then I believe it can quietly create multiple instances of the same library and cause all sorts of havoc. I've seen this with much simpler libraries, and can only begin to imagine how wrong rpi_ws281x might go in this case.

There are ways around it for development, but if you ever plan to deploy as anything more than a single-threaded manual run of your finished Python script then you'll only run into more trouble.

Long story short, building hardware access directly into a Flask application like this is simply the wrong approach. *Especially* in the case of rpi_ws281x, where you need to run your web application as root for it to work at all!

The right approach is to build a separate daemon or service to mediate access to the rpi_ws281x library, and then use a (preferably non-blocking) protocol to transfer commands from the web front-end, to your back-end daemon.

You could do this with anything from a UNIX socket, to web sockets, MQTT, ZeroMQ etc.

This pattern crops up a lot in the Pi world- it's on my TODO list to write an example that shows how to do it.

closed 2018-03-06 by Gadgetoid