jgarff/rpi_ws281x #144: Initialization of two LED strings with C++

litronics, 2016-12-18:
Hello,
thanks for the great library and effort you put into this.
Actually it helped a lot but now I am struggeling to initialize two ledstrings using C++.

Therefore it would be nice if you could translate the following initialization part to C++ coding:

```
ws2811_t ledstring = 
{
    .freq = TARGET_FREQ,
    .dmanum = DMA,
    .channel =
    {
        [0] =
        {
            .gpionum = GPIO_PIN,
            .count = LED_COUNT,
            .invert = 0,
            .brightness = 255,
            .strip_type = STRIP_TYPE,
        },
        [1] =
        {
            .gpionum = 0,
            .count = 0,
            .invert = 0,
            .brightness = 0,
        },
    },
};
```

I already got it for a single channel - however I would like to use the second channel now but struggling there.

Here is what I got so far:
			

```
ws2811_t ledstring =
{
	0,				// device - set automatically
	0,				// rpi_hw - set automatically
	TARGET_FREQ,			// frequency
	DMA,				// DMA num
	GPIO_PIN,			// GPIO-Pin
	0,				// invert
	LED_COUNT,			// LED count
	125,				// brightness
	STRIP_TYPE,			// strip type
	0				// channel
};
```

Thank you very much for your help!

SlySven, 2017-01-26:
Assuming reasonably modern C/C++ (?) set up I wonder whether you can get away with:
```
ws2811_t ledstring[] =
{{ // Data for first instance - assessable as ledstring[0]
	0,				// device - set automatically
	0,				// rpi_hw - set automatically
	TARGET_FREQ,			// frequency
	DMA,				// DMA num
	GPIO_PIN,			// GPIO-Pin
	0,				// invert
	LED_COUNT,			// LED count
	125,				// brightness
	STRIP_TYPE,			// strip type
	0				// channel
},
{ // Data for second instance - assessable as ledstring[1]
	0,				// device - set automatically
	0,				// rpi_hw - set automatically but would be the same as the first one?
	TARGET_FREQ,			// frequency - I suspect you may have to use the same frequency as the first one
	DMA,				// DMA num - dunno about this one
	GPIO_PIN,			// GPIO-Pin - this WILL be a different one to the first one!
	0,				// invert
	LED_COUNT,			// LED count
	125,				// brightness
	STRIP_TYPE,			// strip type
	0				// channel
}};
````

This is _pure_ guesswork on my part... :stuck_out_tongue_winking_eye:

oedze, 2018-09-20:
Well, the issue is still open, i can't seem to get two led strips to work(i actually need 4)

I tried using the second channel in the example, no succes, it will take the color values from the first strip to the second strip. 

When duplicating the ledstring variable to a second one, and initializing that one as wel, it doesn't work either, now all the values from ledstring1 are copied to ledstring 2

Is there a reason for this? if so, how would i fix it?

Gadgetoid, 2022-01-25:
If anyone manages to get a working example for this, it would be lovely to have it added to an examples/ via pull request.

Some basic C examples seem to be conspicuously absent from this library.