jgarff/rpi_ws281x #146: LED string  very DIM

2812user, 2017-01-06:
Hello,
First off, Thanks for creating this library . I was able to get things going pretty easily without delving much into code.

I have some led strips- MCM -25-7522  (says its a SK6812 RGB) that Im trying to get to work correctly
http://www.mcmelectronics.com/content/ProductData/Spec%20Sheets/25-7522.pdf

Setup _  Model B Pi - Transistor Level Shifter - Led Rope . Twin 5V Supplies . Both are grounded properly.
Raspian - Tried with  Latest  & 03/18/16 

I have had to disable the HDMI audio via Config (prior to this strandtest would blink leds randomly at full power) .

CURRENT issue ..  
1. I have level_invert to 0 even though i have a NPN level shifter . Setting to 1 displays nothing on Leds .  
2. Led_brightness is at 255 but LED's are very dim(Without forcing the AUDIO off ,the leds would flash at full brightness,this is why I think correctly working colorwipe is dim) . If I change brightness values to 63 /127  then  completely dark . Randomly one of the LEDS will flash at full brightness during strandtest
3.Theater and Rainbow modes don't work correctly.

sschroe, 2017-01-09:
> Randomly one of the LEDS will flash at full brightness during strandtest

That suggests that the data signal might not be coming through properly. A few things you could try:
* Power LED's at 3.3V and remove the level shifter to see if that works properly (no flickering, better brightness scaling).
* If you have an arduino or other µC laying around try running the LED's with it replacing the rpi.

penfold42, 2017-01-09:
I've seen transistor level shifters do horrible things to the waveform.
The rise time is often too slow so longer data pulses make it but shorter pulses come trough as too short.

Do you have some 74hct (not 74hc) chips lying around ? These make great level shifters

2812user, 2017-01-10:
@sschroe  I will scale down the LEDS to a lower voltage and see if its any different . Dont have any arduinos handy ,might get one to test this exact scenario. What throws me is  parts of strandtest seeming to work fine .

@penfold42  The  transistor is 2n3904 ,should have enough BW to do the trick . Im going to add another shifter in series to see if that helps .  Im thrown by the Led_invert  though ..All Transistor theory and code says that the led_invert should be set to 1 in my use case ,but that completely fails .If I set to 0  (and use the NPN invert) then first few bits of strandtest seem to work.

In this thread, they recommend a few more  blacklists , going to try them .
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=151460

Create a kernel module blacklist file that prevents all the sound drivers from loading:

1. sudo vi /etc/modprobe.d/blacklist-rgb-matrix.conf
2. add the following lines:

blacklist snd_bcm2835
blacklist snd_pcm
blacklist snd_timer
blacklist snd_pcsp
blacklist snd

3. Save the file and quit vi
4. sudo update-initramfs -u
5. reboot and confirm no "snd" modules are running by executing the command "lsmod"

Now give your code a try and see if that helps...

penfold42, 2017-01-10:
Bandwidth isn't the only parameter.

As you're switching you need to look at delay and storage times.

Assuming an open collector inverter the rise times are always going to be terrible because of the pull up resistor and the cable and transistor capacitance.

An HCT is the simplest way to get a nice push pull output stage combined with 3.3v and 5v input thresholds

closed 2017-09-16 by jgarff