jgarff/rpi_ws281x #152: Library does not work with WS2813 LEDs

RHansmann, 2017-02-08:
Jeremy, thanks for this awesome library. But now I have a little problem with WS2813 LEDs:

As far as i figured out in the internet, the WS2813 LED uses reset time >= 250 uS instead of <= 80 uS for WS2812. Suggested are 300 uS. According to several discussions on the net the WS2813 works with this time.

Can somebody modify this library so that it deals with both times depeneding on specified LED type? I figured out that the time is used (only?) in file WS2811.c

I am not  familiar enough with C++, so I ask if somebody can do the required changes to this library  so it will work with both LED.

Thank You very much in advance for Your support.

Best regards
Reinhard

sschroe, 2017-02-10:
If the reset time is the only difference it shouldn't be much of an problem i guess. Is there any decent resource available to buy these yet for testing?

sschroe, 2017-02-26:
I finally got some WS2813's myself and have been playing around with them. For the majority of cases they work just fine already. As soon as you have 10 leds or more just writing the colors to memory takes long enough for the reset to occur. My strip started to behave at around 5 leds. Note that this also is only when hammering out the signals as fast as possible, if you limit the frame rate it works regardless. Due to this reason the included test binary should work fine.

So basically these leds should already work for just about any normal use case. I'll look into making a PR that handles the corner cases as well.

If you are still having issues with them please post further details on your test setup.

jgarff, 2017-09-16:
Should be fixed in latest mainline.  Closing ticket.

closed 2017-09-16 by jgarff