jgarff/rpi_ws281x #154: /dev/mem vs /dev/gpiomem

jgostylo, 2017-02-15:
I am new to raspberry pi so I am not sure about the history of raspbian.

When I build the rpi_ws281x project and try to run strandtest.py I get a permissions error saying that it cannot access /dev/mem.

Ok, so I run `sudo python strandtest.py` and this time there is another error saying it cannot access the GPIO.

More research and it looks like raspbian has separated GPIO out into /dev/gpiomem.  I do a quick grep into the code and find in mailbox.c:
```
    mem_fd = open("/dev/mem", O_RDWR | O_SYNC);
    if (mem_fd < 0) {
       perror("Can't open /dev/mem");
       return NULL;
    }
```
And this is the only literal reference to it.

So I change the "offending" lines to /dev/gpiomem.  I delete the library and the .o files and I compile with scons delete the .egg file from the python libs and rebuild/deploy the .egg file using setup.py again.

When I try to run strandtest.py I get the same permissions error about /dev/mem.

Am I going about this the correct way?  Is there somewhere else in the code that needs to be changed so it tries to access /dev/gpiomem?  It feels to me like my changes are not being packaged but I don't know how that is possible considering the steps I went through to clean things out.

sschroe, 2017-02-26:
Just grabbed Raspbian 2017-01-11 from https://www.raspberrypi.org/downloads/raspbian/ and did a quick test. The C and python examples both work as expected without any further changes.

Does /dev/mem actually exist on your system? Does the C example work for you? Since you mentioned some GPIO error when running it with sudo the problem likely isn't /dev/mem. Please try it with a clean clone of the library and use the commands from the README's to build the weird python thingies.

```bash
git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
scons
sudo apt-get install python-dev swig
python ./setup.py build
sudo PYTHONPATH=".:build/lib.linux-armv7l-2.7" python examples/strandtest.py
```

closed 2017-09-16 by jgarff

Frodox, 2017-12-03:
@jgarff Is it possible to run script without root acces? Just by using `/dev/gpiomem` and so on?
Can you provide/link description why root is strongly needed ?

Because
```
crw-rw---- 1 root gpio 244, 0 Dec  3 18:32 /dev/gpiomem
```
and https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root