jgarff/rpi_ws281x #155: RuntimeError: ws2811_init failed with code -5

Chris1231, 2017-02-20:
can someone help me?

Traceback (most recent call last):
  File "/home/pi/Desktop/examples/strandtest.py", line 83, in <module>
    strip.begin()
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 115, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)

thx
Chris

jgostylo, 2017-02-21:
Do you see another message above that says "Permission denied /dev/mem"?  If so it is possible that you can run with `sudo` to give permissions to /dev/mem but if you have the same issue as me (issue #18) then all this will do is throw another error because raspbian maps the gpio under /dev/gpiomem on my distribution.

Chris1231, 2017-02-21:
I use python shell 2.7.9
Other modules work without problems

sschroe, 2017-02-26:
Which OS and version are you using and which commands did you run? Please provide some reproducable details.

sunnydayw, 2017-03-22:
I'm having the same problem, fresh burn [Raspbian OS image](https://www.raspberrypi.org/downloads/raspbian/) from raspberrypi.org RASPBIAN JESSIE WITH PIXEL. and new raspberrypi zero w board from Adafruit.

Traceback (most recent call last):
  File "strandtest.py", line 83, in <module>
    strip.begin()
  File "build/bdist.linux-armv6l/egg/neopixel.py", line 115, in begin
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

Any solutions?
Thanks,
Chen

shaungordon, 2017-03-22:
@sunnydayw your issue is because RPi Zero W support is not merged in yet.  Easy to fix, see https://github.com/jgarff/rpi_ws281x/issues/162.

sunnydayw, 2017-03-23:
@shaungordon Thank you, I will try it out!

FrankJin-SJ, 2017-04-03:
I have exactly same error message as Chris1231. 

I followed the instruction as https://cdn-learn.adafruit.com/downloads/pdf/neopixels-on-raspberry-pi.pdf

HW: Raspberry Pi 1,   NeoPixel strip 60 LEDs.  

Changed:  LED_COUNT = 60, since I have 60 LED in the strip.  Leave the rest not touched. 

Under Python 2.7 IDE,  F5, run,  error like this

raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)

But if I type :

sudo python strandtest.py

No error,  but the LED lights up very strange, not in control mode. 

Any idea what could be wrong?

Thanks.

penfold42, 2017-04-04:
You need to run it as root like you did with sudo

Have you disabled analog audio ?

FrankJin-SJ, 2017-04-05:
great, it works after I disable the audio.  Thanks a lot!

mmgill, 2017-08-15:
I have the same issue but cant figure it out..

I have a test .py script that just lights up an LED.  Work fine if I use "sudo python test.py" but not when using "python3 test.py".  I get the "RuntimeError: ws2811_init failed with code -5 (mmap() failed)" error.

I am trying to add an LED into the google AIY main.py script which is executed using "python3 main.py"  and getting the same RuntimeError.  Is there any way to get around using sudo?  Trying "sudo python main.py" brings me into a whole other mess of import errors..

JAICHANGPARK, 2017-11-17:
I have Same problem,,
How to solve this?

Gadgetoid, 2018-03-06:
If you're using PWM/PCM there is no way to avoid running as root.

You may have some luck with SPI on GPIO 10.

closed 2018-03-06 by Gadgetoid

palorisisis, 2022-04-17:
> can someone help me?
> 
> Traceback (most recent call last): File "/home/pi/Desktop/examples/strandtest.py", line 83, in strip.begin() File "build/bdist.linux-armv7l/egg/neopixel.py", line 115, in begin raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, message)) RuntimeError: ws2811_init failed with code -5 (mmap() failed)
> 
> thx Chris

is simple,you need add sudo!