jgarff/rpi_ws281x #159: How many LEDs can you drive?

ArosPrince, 2017-03-02:
Hi, first of all thanks for the library. I have tried it and so far I must say it's great. There is one thing I am missing though... How many LEDs can you drive? What's the theoretical maximum? I expected to find the information somewhere in the readme, but it's not there...

Thanks!

sschroe, 2017-03-02:
See #8, basically something up to 3000 should be no problem.

ArosPrince, 2017-03-06:
Thanks! And I guess at that count the refresh rate would be only about 11 Hz, right? Practically a bit lower because of the CPU overhead...

sschroe, 2017-03-06:
11 Hz should be right, somewhat depends on the LED type though (RGBW ones are slower). Also if you make use of both PWM channels you should be able to double that (haven't tried it yet).

marcuszerep, 2022-06-27:
FYI to anyone else reading this...

Real world example: Pi using single PWM >> level shifter (https://learn.adafruit.com/neopixels-on-raspberry-pi/raspberry-pi-wiring) >> 15 foot double shielded RCA (for signal) >> to 2232 LEDs (WS2815)

Without the level shifter and double shielding signal cable, we did experience signal degradation which led to flickering the whole set of LEDs--color shift, 1 LED signal off. Refresh rate is a bit slow with the signal size but fine.