jgarff/rpi_ws281x #160: Orange Pi Zero support ?

Adelysnet, 2017-03-10:
Any chance to make it works on a orange pi zero running Armbian Jessie ?

For what its worth, I've tried to add hardware support in rpihw.c, copying the RPI Zero part and changing the revision to 0x0000 (from my cpuinfo file). It fails with a device error, the vcio file does not exist : 

> Can't open device file
> : No such device or address
> ws2811_init failed: Failed to create mailbox device

sschroe, 2017-03-10:
Its not going to work just like that as the Orange Pi uses a different SoC. If the used AllWinner H2 SoC supports PWM with DMA then it should be possible to make the library work but this would require access to the documentation so we can figure out how to use it.

closed 2017-03-24 by jgarff

cpwtech, 2017-03-25:
What documentation would you need I have a pdf Datasheet on the Allwinner H3 that is like 614 pgs long.