jgarff/rpi_ws281x #161: Flickering / Random blinking Neopixels, Rapsberry Pi 3

DonRichie, 2017-03-10:
Hello,

I am trying to set up a matrix of 9 Neopixel. 
They work fine when using the Arduino Uno, but unfortunately not when I use rpi_ws281x on a Raspberry Pi 3 with Raspian.

I made a demo video to show you what happens:
[https://s.ricl.de/ohnoes.mp4](https://s.ricl.de/ohnoes.mp4)

Basically with Arduino Uno the expected behavior occurs: The LEDs flah nicely in red, green and blue.
On Raspberry Pi the LEDs are flickering random and weird.

The Arduino Uno C program I use is an example from the Adafruit Page:
```
#include <Adafruit_NeoPixel.h>
 
#define PIN      6
#define N_LEDS 12
 
Adafruit_NeoPixel strip = Adafruit_NeoPixel(N_LEDS, PIN, NEO_GRB + NEO_KHZ800);
 
void setup() {
  strip.begin();
}

void loop() {
  chase(strip.Color(255, 0, 0)); // Red
  chase(strip.Color(0, 255, 0)); // Green
  chase(strip.Color(0, 0, 255)); // Blue
} 

static void chase(uint32_t c) {
  for(uint16_t i=0; i<strip.numPixels()+4; i++) {
      strip.setPixelColor(i  , c); // Draw new pixel
      strip.setPixelColor(i-4, 0); // Erase pixel a few steps back
      strip.show();
      delay(25);
  }
}
```

On Raspberry Pi 3 I am using "strandtest.py" and changed nothing expect the LED Count in the configuration header:
```
# LED strip configuration:
LED_COUNT      = 9 # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
```

My wiring is as follows for both Devices:
[https://cdn-learn.adafruit.com/assets/assets/000/030/892/medium800/leds_Wiring-Diagram.png](https://cdn-learn.adafruit.com/assets/assets/000/030/892/medium800/leds_Wiring-Diagram.png?1456961114)

Does somebody have an idea why I fail to make it work on Raspberry?

Regards

Richard

sschroe, 2017-03-10:
You need a level shifter for the data input to convert the RPI's 3.3V to 5V levels. Also make sure that the audio module is blacklisted as mentioned in the readme.

DonRichie, 2017-03-11:
Hello, 

thanks for your answer.

I fortunately have the 74AHCT125 from Adafruit. It is a "Quad Level-Shifter (3V to 5V)".

Now I tried to set up like in the following image:
[https://cdn-learn.adafruit.com/assets/assets/000/019/658/medium800/leds_levelshifter_bb.png?1410393346](https://cdn-learn.adafruit.com/assets/assets/000/019/658/medium800/leds_levelshifter_bb.png?1410393346)
I added the recommended resistor on the data pin and the capacitor like in the previous example.

Unfortunately my result is the same. The leds are randomly flashing.

Here is a photo and another video:
[https://s.ricl.de/withshifter.mp4](https://s.ricl.de/withshifter.mp4)
[https://s.ricl.de/withshifter.jpg](https://s.ricl.de/withshifter.jpg)

------------------------------------------

After writing the upper part I tried to blacklist the kernel module. 
And amazing: It works now (nearly perfect)!

By the way: I tried it out. I can omit the 74AHCT125  Level Shifter and everything works the same.

[https://s.ricl.de/solution.mp4](https://s.ricl.de/solution.mp4)

Nearly perfect because one LED blinks white sometimes. My config is definately saying it should enlighten 12 LEDs.  
I made another video and slowed everything down to a 200ms delay to show you:
[https://s.ricl.de/missingpixel.mp4](https://s.ricl.de/missingpixel.mp4)
The camera didnt catch it perfectly, but you should see on a lower left pixel. In reality it is clearly visible white.

As you can see the problem occurs randomly, but mostly during a color wipe. What could that be? 
Ist it just a boken LED or something else?

Regards

Richard

sschroe, 2017-03-11:
While it might work without the level shifter there is no guarantee for that, this somewhat depends on the tolerances of the LEDs and such. So the proper way to drive them is with the shifter in place, otherwise you are working outside of the specification. So be sure to keep the shifter in place, otherwise this could result in things like those random flashes you get now, especially when the strips get longer.

Whether this is a broken led or not you should be able to test with your Arduino, if it works there it should be alright.

electrokean, 2017-03-11:
Indeed, the level shifter is needed for reliable operation.  See recent comments on https://github.com/jgarff/rpi_ws281x/issues/103 relating to this.

DonRichie, 2017-03-11:
I tested the LEDs with my Arduino in several ways.  Fast flashing, Slow flashing, high brightness, low brightness and everything else coming to my mind.
Everything works fine and no LED has a wrong color at any time.

The behavior that one specific LED does a mistake in showing its color happens only when I use the Raspberry Pi 3.  This includes when I use the level shifter.
So I think the Raspberry might have some accuracy problems or something like that. Regarding the level shifter I tend to say "If it works without, it works". But of course the situation might differ when I use longer LED strips.

Thanks again for giving me the hint to blacklist the kernel module. I kind of overread it in the Readme. Next thing to try is using my ESP8266 to enlighten the LEDs. I close the issue for now.

closed 2017-03-11 by DonRichie