jgarff/rpi_ws281x #162: Code not running on Pi Zero W

Heliosphan, 2017-03-11:
Hi,
Much appreciation towards the authors of this library, awesome work.
Another year, another Pi, the code now no longer runs with the new Raspberry Pi Zero W.

Error - "ws2811_init failed: Hardware revision is not supported"

I had a quick root around, found that there was no definition for the Zero W in the raspi_hw.c code, but when I added a new one, and changed the hardware revision number to the value /proc/cpuinfo has 0x9000c1, the code runs, but doesn't light any of my WS2812s.
Code to define the Pi W looks like - 
    {
        .hwver  = 0x9000c1,
        .type = RPI_HWVER_TYPE_PI2,
        .periph_base = PERIPH_BASE_RPI2,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi Zero W",
    },
I tried using the Pi HWVER 1 defines instead of these PI 2 defines, and I get the same result. But with the Pi 2 one the program refuses to Ctrl-C, so I'd imagine it's distinctly unhappy with the Pi 2 definitions being used.

My /proc/cpuinfo - 

root@zerow1:~/rpi_ws281x# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2708
Revision        : 9000c1
Serial          : 00000000819?????

An update to this by somebody more competent than me would be much appreciated.

Heliosphan, 2017-03-11:
OK I got the LED matrix lit up, but only full white with hints of colour in there, with a weird flicker. It's pretty consistent, happened 5 times in a row when resetting the matrix and running the program again.
Code looks like - 

    {
        .hwver  = 0x9000c1,
        .type = RPI_HWVER_TYPE_PI1,
        .periph_base = PERIPH_BASE_RPI,
        .videocore_base = VIDEOCORE_BASE_RPI,
        .desc = "Pi Zero W",
    },

The original failure may have been down to poor wiring.

duquesne419, 2017-03-21:
I just wanted to check in to see if there was any update on this? I'm completely new to coding, so am not likely any help, but am willing to try.

shaungordon, 2017-03-21:
I tested the fix above last night and it works perfectly.  Looks like there are already two pull requests for this, so hopefully it will be merged soon.

Heliosphan, 2017-03-21:
Interesting, thanks shaungordon - looks like I'll have to take another look at my circuit layout as I couldn't for love nor money get my strip to light with this code!

shaungordon, 2017-03-21:
FWIW, I am powering 15 pixels using a single 18650 battery, so lower than 5v.  I did also get it working using a diode to drop down 5v usb as shown on the adafruit site.  Feel free to email me if you want more info.

Heliosphan, 2017-03-21:
Yup, ok, I built a bad circuit.
I got it working with an 8x8 matrix. 
In the past I've built two different Pi driven WS281x matrices. 
I agree, I much prefer under-driving the LEDs with 4.2v, so no logic level conversion is required - I've never been able to drive 3v3 to 5v directly, without conversion and I doubt it's even possible. 

I've no idea what I've been doing wrong this time round though! For some reason it only works now with a 280ohm resistor between GPIO18 and LED data in! I've never needed to do this before!

Thanks for the confirmation.

duquesne419, 2017-03-26:
I've added the description of the Pi Zero W to rpi_hw.c, but I'm still getting the hardware error. Is there somewhere else I need to be updating, or is it just rpi_hw.c?

Heliosphan, 2017-03-26:
This could suggest my Pi Zero W has a different hardware revision ID from yours. Run cat /proc/cpuinfo and note the revision number. It's it different to 9000c1?
If so change the hwver value to yours with 0x... At the start.

SergejMoor, 2017-03-30:
Hey guys,

i ran into another Problem, that i had to fix before the library would properly work with my pi zero w:
Terminal -> sudo raspi-config -> 7 Advanced Options -> A4 Audio -> 2 Force HDMI
fixed my Problem. My LED String was no behaving right (flickering, wrong number of LEDs) until i forced Audio to HDMI.

pinheadmz, 2017-08-08:
I'm getting this error. RPi Zero W. After installing the latest code via git, the `sudo ./test` command lights up my neopixels nicely! But all of the examples like `sudo python python/examples/strandtest.py ` return the error:

```
Traceback (most recent call last):
  File "strandtest.py", line 87, in <module>
    strip.begin()
  File "build/bdist.linux-armv6l/egg/neopixel.py", line 115, in begin
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

pinheadmz, 2017-08-08:
Never mind!
I never ran the command to install the python wrapper:
`sudo python setup.py install`
Thanks :-)

closed 2018-03-06 by Gadgetoid