jgarff/rpi_ws281x #165: multistrand problem

fusionfoto, 2017-03-24:
First, thanks for all of the great work you've done on this library. 

In single-thread mode, this is rock solid... I've driven over 720 WS2811 RGBW LEDs without issue.

I decided to break up some of those LEDs into multistrand, so I have 144 RGB on GPIO 13 and 144 RGB GPIO on 18. 

I wrote a simple Kr function and put it into threads in python built specifically inside a local copy of multistrand.py

The significant code looks like this:

# NeoPixel library strandtest example
# Author: Tony DiCola (tony@tonydicola.com)
#
# Direct port of the Arduino NeoPixel library strandtest example.  Showcases
# various animations on a strip of NeoPixels.
import time

from neopixel import *
from threading import Thread

# LED strip configuration:
LED_1_COUNT      = 144      # Number of LED pixels.
LED_1_PIN        = 18      # GPIO pin connected to the pixels (must support PWM! GPIO 13 and 18 on RPi 3).
LED_1_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_1_DMA        = 5       # DMA channel to use for generating signal (Between 1 and 14)
LED_1_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_1_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_1_CHANNEL    = 0       # 0 or 1
#LED_1_STRIP      = ws.SK6812_STRIP_GRBW
LED_1_STRIP     = ws.WS2811_STRIP_GRB

LED_2_COUNT      = 144      # Number of LED pixels.
LED_2_PIN        = 13      # GPIO pin connected to the pixels (must support PWM! GPIO 13 or 18 on RPi 3).
LED_2_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_2_DMA        = 10      # DMA channel to use for generating signal (Between 1 and 14)
LED_2_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_2_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_2_CHANNEL    = 1       # 0 or 1
LED_2_STRIP      = ws.WS2811_STRIP_GRB

# Main program logic follows:
if __name__ == '__main__':
        # Create NeoPixel objects with appropriate configuration for each strip.
        strip1 = Adafruit_NeoPixel(LED_1_COUNT, LED_1_PIN, LED_1_FREQ_HZ, LED_1_DMA, LED_1_INVERT, LED_1_BRIGHTNESS, LED_1_CHANNEL, LED_1_STRIP)
        strip2 = Adafruit_NeoPixel(LED_2_COUNT, LED_2_PIN, LED_2_FREQ_HZ, LED_2_DMA, LED_2_INVERT, LED_2_BRIGHTNESS, LED_2_CHANNEL, LED_2_STRIP)

        # Intialize the library (must be called once before other functions).
        strip1.begin()
        strip2.begin()

        print ('Press Ctrl-C to quit.')

        # Black out any LEDs that may be still on for the last run
        blackout(strip1)
        blackout(strip2)

        t=Thread(target=Kr, args=(strip1,))
        t.start()
        t=Thread(target=Kr, args=(strip2,))
        t.start()

-----

The code works, so that's not really the issue. The problem is after its run about 6 passes, the LEDs lose sync, and color integrity (first in one string, then both) The string that loses integrity starts going faster than the one that hasn't lost integrity.  If it matters, its the GPIO13 strand that loses integrity first.  By about pass 10 both are well off the reservation.

This doesn't happen when driving more than 288 LEDs on a single channel.  There is no other load on the RPi, and individually the strings work fine. 

If I use a default function (like multicolorWipe, the same behavior occurs. 

My guess is that the PWM emulation is losing something along the way... perhaps I'm not pausing long enough (about 0.10 s per pixel move currently). Is there someway I should be re-disciplining the clock along the way?

thanks in advance!

fusionfoto, 2017-03-24:
If I comment out either thread, but otherwise leave the code unmodified, no errors occur for at least 20 passes that I've watched.

sschroe, 2017-03-25:
For this library you should only initialize a single instance and only a single thread for the library calls. Simultanous rendering of the two channels is handled internally.

fusionfoto, 2017-03-27:
Hi, 

So I initialize the strings outside of the threads. Because I eventually want the two strings to do totally different things, I figured threads, or even separate processes would make the coding cleaner.

I can try putting all of the commands into the same thread/process, but I can't see why issuing the commands via separate processes or separate threads would mess up the library. I wouldn't mind making the library threadsafe if I knew what was getting hosed. 

Thanks!

sschroe, 2017-03-27:
The PWM hardware to do the work only exists once on the Pi, so multi threading can't do anything there. You just tell it to start generating the signal for both channels and it does its thing.

Besides that the rendering speed is mostly limited by the ws2811 protocol itself which takes something like 30µs per led. For 144 leds that already is 4.3ms. Due to this the time it takes for the CPU to prepare the colors is negligible. Also the library already allows you to prepare the next set of colors while the previous one is being rendered.

Or basically there is nothing to gain by making the library multithreaded.

fusionfoto, 2017-03-27:
ok, I get all of that, so I revised the script to run without threads, update the two strings, then do a show on each one, then pause, and move on... 

That freaked out even worse than the threads.

But given your feedback... that it should "just work"... I tried using the same DMA channel... (5 in this case) and that seemed to unscrew-up everything.  The threaded version works great on a shared DMA channel 10... 

I thought each DMA channel was separate, and since we are communicating to different strings with different memory, that DMA channel shouldn't matter much less prefer to be shared.  However, both the single and multi-threaded versions of my code now work beautifully. 

thanks for your help and insight.

closed 2017-03-27 by fusionfoto