jgarff/rpi_ws281x #166: Control RGBW Pixels oder network

codmpm, 2017-03-27:
First, thank you for your work. I write here cause I don't really know where to ask else. So see this more as a guidance and tip issue :-)

I have a rather large WS2812-RGBW installation with 1608 Pixels in one strip. I've planed to write own software for this, which did work. Currently I can switch the pixel strip to a solid self-defined color via network with an Arduino Mega and an Ethernet-Shield. 
Obviously the Mega has not enough horsepower to do the heavy liftig for so many pixels. So the logical next step is an RPi. This is how I found your library. I already did some tests with WS2811-RGB Pixels and the 74AHCT125 which worked.

But I have to admit, that I can't get my head arround the animation part. So I'm willing to buy an appropriate controller for this but I am not able to find anything which can be controlled over network.

As I'm kind of stuck, probably you can recommend some software, maybe using your library, for the RPi which can be controlled via network? Something where I can predefine animations and "load" them via a request.

Would be great to get some input.

Cheers, 
Patrik

codmpm, 2017-03-27:
Thank you for your response :-)

I've found falcon christmas which seems to be a made for this but can't find a way to control it other than with an external player. Maybe I've missed something...
As far as I unserstand the animations could be built using some kind of software (Lighting Sculp) and be placed somewhere.

On Aliexpress there are several Players that can be controlled with an iOs/Android App but are limited to 1024 Pixels and can't handle RGBW... https://de.aliexpress.com/item/led-WiFi-controller-1-port-control-max-2048-pixels-support-WS2811-WS2812-DMX512-etc-Controlled-by/32507572159.html for example.
I also thought about some ARTNet WS2812 bridge... but the don't support RGBW either.

The node-red NeoPixel Node also does not support RGBW. I think I should have bought RGB ones :-P

Think you understand why I got stuck ;-)

sschroe, 2017-03-27:
Well you wouldn't have to bitbang the protocol, thats what this library is for. You only tell the library which leds to set to which color and it handles the rest for you. Also there is python bindings available so you can get stuff working relatively quickly with that, even the networking stuff.

codmpm, 2017-03-27:
A friend of mine is fluent in python... I already asked him... maybe he could spare some time 👍

closed 2017-09-16 by jgarff