jgarff/rpi_ws281x #167: Scons fails to run on Rpi3(Arch Linux)

C0rn3j, 2017-03-29:
Steps to reproduce

git clone https://github.com/jgarff/rpi_ws281x.git
cd rpi_ws281x
scons

error: 

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
Version version.h
CC      main.o
CC      mailbox.o
mailbox.c: In function 'mbox_open':
mailbox.c:275:13: error: In the GNU C Library, "makedev" is defined
 by <sys/sysmacros.h>. For historical compatibility, it is
 currently defined by <sys/types.h> as well, but we plan to
 remove this soon. To use "makedev", include <sys/sysmacros.h>
 directly. If you did not intend to use a system-defined macro
 "makedev", you should undefine it after including <sys/types.h>. [-Werror]
     if (mknod(filename, S_IFCHR|0600, makedev(100, 0)) < 0) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
cc1: all warnings being treated as errors
scons: *** [mailbox.o] Error 1
scons: building terminated because of errors.

penfold42, 2017-05-17:
Did either of you try changing the include as suggested by the warning ?

ThePendulum, 2017-05-18:
I did, to no avail. I eventually built it by just crudely removing the -Werror flag from linux.py.

C0rn3j, 2017-05-19:
>did you ever end up finding a solution?

I did I think that's why there's an EDIT.

My LED strip is dead though so can't test.

jgarff, 2017-09-16:
Closing due to inactivity.  Please re-open if necessary.  Thanks.

closed 2017-09-16 by jgarff