jgarff/rpi_ws281x #168: Replace Pi detection with more generic approach

penfold42, 2017-04-01:
Using this approach should remove the need to constantly add new boards and revisions.

https://gist.github.com/petarov/61bc90a8b0d27c8f7e846ff81871dfa7

Piblaster has some code we could borrow

Gadgetoid, 2017-04-03:
I suggested a similar approach a while back, which was countered with some sound reasoning but frankly I'm not sure it's the lesser of two evils to be requiring explicit device support. The Pi has shown no sign of changing significantly in 5 years, and yet we're opting to break rpi_ws281x with every new release for, as far as I can tell, absolutely no gains whatsoever.

See: https://github.com/jgarff/rpi_ws281x/pull/104/files

Gadgetoid, 2017-04-11:
Well, having tested my old code on a couple of Pi's it failed on my dev Pi 3 due to the lack of `/proc/device-tree/axi/vc_mem/reg`, I don't know why this node is missing but it doesn't make my solution particularly dependable!

The PiBlaster code you mention is, presumably, the following:

```c
static void
get_model(unsigned mbox_board_rev)
{
	int board_model = 0;

	if ((mbox_board_rev & BOARD_REVISION_SCHEME_MASK) == BOARD_REVISION_SCHEME_NEW) {
		if ((mbox_board_rev & BOARD_REVISION_TYPE_MASK) == BOARD_REVISION_TYPE_PI2_B) {
			board_model = 2;
		} else if ((mbox_board_rev & BOARD_REVISION_TYPE_MASK) == BOARD_REVISION_TYPE_PI3_B) {
			board_model = 3;
		} else if ((mbox_board_rev & BOARD_REVISION_TYPE_MASK) == BOARD_REVISION_TYPE_CM3) {
			board_model = 3;
		} else {
			// no Pi 2, we assume a Pi 1
			board_model = 1;
		}
	} else {
		// if revision scheme is old, we assume a Pi 1
		board_model = 1;
	}

	switch(board_model) {
		case 1:
			periph_virt_base = 0x20000000;
			periph_phys_base = 0x7e000000;
			mem_flag         = MEM_FLAG_L1_NONALLOCATING | MEM_FLAG_ZERO;
			break;
		case 2:
		case 3:
			periph_virt_base = 0x3f000000;
			periph_phys_base = 0x7e000000;
			mem_flag         = MEM_FLAG_L1_NONALLOCATING | MEM_FLAG_ZERO; 
			break;
		default:
			fatal("Unable to detect Board Model from board revision: %#x", mbox_board_rev);
			break;
	}
}
```

But seems also to depend upon:

```c
unsigned get_board_revision(int file_desc)
{
   int i=0;
   unsigned p[32];
   p[i++] = 0; // size
   p[i++] = 0x00000000; // process request

   p[i++] = 0x10002; // (the tag id)
   p[i++] = 4; // (size of the buffer)
   p[i++] = 0; // (size of the data)
   p[i++] = 0; // response buffer

   p[i++] = 0x00000000; // end tag
   p[0] = i*sizeof *p; // actual size
   
   mbox_property(file_desc, p);
//    printf("get_board_revision mbox response buffer:\n");
//    int pi = 0;
//    int length = p[0]/sizeof *p;
//   for (; pi < length; pi++) {
//        printf("%d: %#x\n", pi, p[pi]);
//   }
   return p[5];
}
```
For now I'm cutting my losses and rolling back to the old method in my fork.

penfold42, 2017-04-11:
That's the code - if I had the time I'd use their approach for anything with a new style version string. 

Do we ever care about the board type ? 
I suspect we really only care about the CPU type so we know where the peripheral addresses are.

penfold42, 2017-05-28:
#183

Gadgetoid, 2019-11-06:
I think it's safe to say we've settled into the pattern of supporting new Pi models as and when they are discovered (usually by this code breaking for a new user) so it's probably time to close this old chestnut and defer any future conversation to the PR.

closed 2019-11-06 by Gadgetoid