jgarff/rpi_ws281x #169: Syncing both PWM channels

lukefanning, 2017-04-07:
Hi, we've built a very large LED wall powered by a RPi 2 Model B running this library (check it out here https://github.com/SolidStateGroup/rio). We use both PWM channels to drive 1020 pixels each. This works OK but what we've noticed is, is that we get noticeable tearing on output such as scrolling text.

Is there any solution to this? I've tried a few things like inserting a delay after calling .show() on both strips and having both .show() functions run in parallel but it didn't really make a difference (in the case of the latter it was actually worse). You can see the python script that we are using here https://github.com/SolidStateGroup/rio/blob/master/rio-server/firmware/rpi/server.py. Thanks

sschroe, 2017-04-07:
I don't use the python stuff myself but from the looks of it it does not support using two channels in parallel for the neopixel class. It should be doable with the lowlevel variant though (examples/lowlevel.py).

For that use `ws.new_ws2811_t()` to get a single instance (and no more) and then use `ws.ws2811_channel_get` to switch between the channels on that instance.

ssg-luke, 2017-04-11:
I finally found some time to try this! It's only rendering the top half of the wall for me. Perhaps there's something wrong with my script? https://github.com/SolidStateGroup/rio/blob/master/rio-server/firmware/rpi/lowlevel-server.py

sschroe, 2017-04-11:
I don't see anything wrong from a quick glance. You could try disabling one channel and test with only the other one being active to see if it makes any difference. Might take a bit of debugging there.

ssg-luke, 2017-04-12:
I messed up the LED index to the 2nd channel. Sorted it out. Works really well! Seems like we're getting a better frame rate as well over the existing script we were using. Thanks for the tip!

closed 2017-04-12 by ssg-luke