jgarff/rpi_ws281x #172: (very) minor documentation issue in python/README.md

gschintgen, 2017-04-12:
Hi, a big thanks for your ws281x libraries!
It worked (almost) right off the bat!

I'd just like to point out that in python/README.md it says to use
PYTHONPATH=".:build/lib.linux-armv7l-2.7"
while it should be "./build...".
It may also be of interest for some users that on an older Pi1 it's actually "armv6l" instead of "armv7l".

penfold42, 2017-05-28:
I dont see anything wrong with the PYTHONPATH.

Colon is a path separator so it's adding both the current dir and the build/lib.linux-armv7l-2.7

We should add the old pi1 path string to it tho... what is the exact folder name that gets created?

Gadgetoid, 2018-03-06:
This is partially correct, the listed path is dependent upon both the architecture and Python version used to build the library. Since it's not a given that "python" refers to "python2.7", it's probably worth fixing.

For the time being, and the sake of simplicity, I'd suggest removing instructions for running the uninstalled version of the library altogether, and sticking with a simple install -> run procedure.

Gadgetoid, 2018-08-29:
All Python issues should now be directed here: https://github.com/rpi-ws281x/rpi-ws281x-python

closed 2018-08-29 by Gadgetoid