jgarff/rpi_ws281x #173: Error build 'ws2811.c': 'cast from pointer to integer of different size'

lukaszgard, 2017-04-25:
Hello,

I'm trying building minimal raspberry pi zero image using Yocto Project, My goal is to create minimal image with script runs by sysinit using this library. I'm not familiar with C but I'm wondering why I get this  (trivial) error during compilation 'ws2811.c' that I may successfully build (whole library) on my another raspberry pi 2 with raspbian OS without any errors. Thanks for every hints and help!

My recipe was stuck trying to build in root (repository) location 'scons':
`sudo scons`

```
ERROR: python-ws2812-1.0+git999-r0 do_compile: Function failed: do_compile (log file is located at /home/astor/Documents/poky/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/python-ws2812/1.0+git999-r0/temp/log.do_compile.11838)
ERROR: Logfile of failure stored in: /home/astor/Documents/poky/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/python-ws2812/1.0+git999-r0/temp/log.do_compile.11838
Log data follows:
| DEBUG: Executing python function externalsrc_compile_prefunc
| NOTE: python-ws2812: compiling from external source tree /home/astor/Documents/poky/rpi-build/workspace/sources/python-ws2812
| DEBUG: Python function externalsrc_compile_prefunc finished
| DEBUG: Executing shell function do_compile
| scons: Reading SConscript files ...
| scons: done reading SConscript files.
| scons: Building targets ...
| CC      ws2811.o
| ws2811.c: In function 'setup_pwm':
| ws2811.c:289:23: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
|      dma_cb->dest_ad = (uint32_t)&((pwm_t *)PWM_PERIPH_PHYS)->fif1;
|                        ^
| cc1: all warnings being treated as errors
| scons: *** [ws2811.o] Error 1
| scons: building terminated because of errors.
| WARNING: /home/astor/Documents/poky/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/python-ws2812/1.0+git999-r0/temp/run.do_compile.11838:1 exit 2 from 'scons'
| ERROR: Function failed: do_compile (log file is located at /home/astor/Documents/poky/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/python-ws2812/1.0+git999-r0/temp/log.do_compile.11838)
```

And this how look like my recipe:
```bitbake
LICENSE = "BSD"
LIC_FILES_CHKSUM = "file://LICENSE;md5=9dcf340793a1d73c5211edc8238767dc"

SRC_URI = "git://github.com/richardghirst/rpi_ws281x.git;protocol=https"

PV = "1.0+git${SRCPV}"
SRCREV = "39afaac5f2b8b307d7d7b5f2f790fbb6759bda5e"

S = "${WORKDIR}/git"

DEPENDS = "swig-native"

inherit scons setuptools 

do_compile_prepend() {
    cd ${S}/
    scons
    ${PYTHON} ${S}/python/setup.py install
}
```

GCC and scons versions:

```gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-11)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

SCons by Steven Knight et al.:
script: v2.5.1.rel_2.5.1:3735:9dc6cee5c168[MODIFIED], 2016/11/03 14:02:02, by bdbaddog on mongodog
engine: v2.5.1.rel_2.5.1:3735:9dc6cee5c168[MODIFIED], 2016/11/03 14:02:02, by bdbaddog on mongodog
engine path: ['//home/astor/Documents/poky/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/python-ws2812/1.0+git999-r0/recipe-sysroot-native/usr/bin/../../usr/lib/python2.7/site-packages/SCons']

sschroe, 2017-04-25:
Try replacing `uint32_t` with `uintptr_t` there.

lukaszgard, 2017-04-25:
Well @sschroe Your advice work well ! 

But there is similar issue in another file:

```| DEBUG: Executing python function externalsrc_compile_prefunc
| NOTE: python-ws2812: compiling from external source tree /home/astor/Documents/poky/rpi-build/workspace/sources/python-ws2812
| DEBUG: Python function externalsrc_compile_prefunc finished
| DEBUG: Executing shell function do_compile
| scons: Reading SConscript files ...
| scons: done reading SConscript files.
| scons: Building targets ...
| CC      ws2811.o
| CC      pwm.o
| CC      dma.o
| CC      board_info.o
| CC      mailbox.o
| mailbox.c: In function 'mapmem':
| mailbox.c:68:33: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
|        printf("mmap error %d\n", (int)mem);
|                                  ^
| cc1: all warnings being treated as errors
| scons: *** [mailbox.o] Error 1
| scons: building terminated because of errors.
```

Did You know and may describe me why this errors comes during installation with Yocto ?

sschroe, 2017-04-25:
Replacing that one with `uintptr_t` should take care of it as well. Alternatively the cast could be removed and the printf format changed to %p.

These errors occur because the casts are not correct. Pointers have varying sizes, on 64 bit systems for example they can be expected to be 64bit. Casting such a 64bit value to `int` or `uint32_t` would result in some data being lost and the results might not be what one expected.

The usual RPI distros are 32 bit based. So if your yocto uses 64 bit that likely is the reason.

rpisesja, 2017-04-28:
I'm so appreciate for you answers and feedback @sschroe ! But I understand that there is no problem with this library but with yocto/scons cross-compilation. Those workarounds let me build 'test' file as a 64Bits file - that will for sure not run on raspberry pi. I will try to continue this 'issue' (not related to this lib) on stack-overflow -> [http://stackoverflow.com/questions/43618735/yocto-raspberry-pi-error-during-installation-rpi-ws281x-library](url)

lukaszgard, 2017-04-28:
I'm so appreciate for you answers and feedback @sschroe ! But I understand that there is no problem with this library but with yocto/scons cross-compilation. Those workarounds let me build 'test' file as a 64Bits file - that will for sure not run on raspberry pi. I will try to continue this 'issue' (not related to this lib) on stack-overflow -> http://stackoverflow.com/questions/43618735/yocto-raspberry-pi-error-during-installation-rpi-ws281x-library

closed 2017-04-28 by lukaszgard