jgarff/rpi_ws281x #176: freeze after rpi-update

SimonJakubowski, 2017-05-08:
After installing the latest rpi firmware my Raspberry Pi 3 is freezing while running ./test

/var/log/syslog
May  8 09:04:24 raspberrypi kernel: [   17.766054] mmc0: timeout waiting for hardware interrupt.
May  8 09:04:24 raspberrypi kernel: [   17.766068] [c0ad0671] PRD2 1 0
May  8 09:04:24 raspberrypi kernel: [   17.766076] [c0ad067d] PRD3 b8f62f84 0
May  8 09:04:24 raspberrypi kernel: [   17.766083] [c0ad067e] PDM> b99d1d0c 0
May  8 09:04:24 raspberrypi kernel: [   17.766090] [c0ad067f] REQ< b99d1c34 10801
May  8 09:04:24 raspberrypi kernel: [   17.766097] [c0ad0683] CMD< 12 30ab0
May  8 09:04:24 raspberrypi kernel: [   17.766103] [c0ad068e] CMDD 8 200
May  8 09:04:24 raspberrypi kernel: [   17.766111] [c0ad0690] SDMA b99d1d0c b99b5e10
May  8 09:04:24 raspberrypi kernel: [   17.766118] [c0ad0696] FCM< b99d1c34 b99d1ca4
May  8 09:04:24 raspberrypi kernel: [   17.766125] [c0ad0698] RSP  900 0
May  8 09:04:24 raspberrypi kernel: [   17.766132] [c0ad0699] FCM> b99d1c34 0
May  8 09:04:24 raspberrypi kernel: [   17.766139] [c0ad069a] CMD  12 200
May  8 09:04:24 raspberrypi kernel: [   17.766146] [c0ad069b] REQ> b99d1c34 0
May  8 09:04:24 raspberrypi kernel: [   17.766153] [00ad09f8] DMA< b99d1d0c 1
May  8 09:04:24 raspberrypi kernel: [   17.766159] [00ad09f9] DMA  52 10902
May  8 09:04:24 raspberrypi kernel: [   17.766166] [00ad09ff] FDA< b99d1c34 0
May  8 09:04:24 raspberrypi kernel: [   17.766173] [00ad0a00] TCM< b99d1d0c 0
May  8 09:04:24 raspberrypi kernel: [   17.766180] [00ad0a01] CMD< c 0
May  8 09:04:24 raspberrypi kernel: [   17.766187] [00ad0a05] FCM< b99d1c34 b99d1cd8
May  8 09:04:24 raspberrypi kernel: [   17.766193] [00ad0a0a] RSP  b00 0
May  8 09:04:24 raspberrypi kernel: [   17.766200] [00ad0a0b] FCM> b99d1c34 b99d1cd8
May  8 09:04:24 raspberrypi kernel: [   17.766207] [00ad0a0c] TCM> b99d1d0c 0
May  8 09:04:24 raspberrypi kernel: [   17.766214] [00ad0a0d] FDA> b99d1c34 b99d1cd8

christophevg, 2017-05-11:
I'm seeing similar behaviour since I installed a fresh RPi image. After running a Neopixel script for some time (less than it had been running before on an older image), the file system becomes read-only, requires a reboot, and the logs are indeed filled with comparable messages as shown above. I'll try to hunt down the older image to have a reference to a working situation. To be continued.

christophevg, 2017-05-11:
I've tested my Neopixel script on a older Raspbian image on a Pi on which I knew that was ok, and it runs smoothly.

This system can be identified as:

```bash
$ uname -a
Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ ls -l /boot/kernel.img 
-rwxr-xr-x 1 root root 3963140 Feb 25  2016 /boot/kernel.img
```

Looking at http://downloads.raspberrypi.org/raspbian/images/ my guess is that this might be: `raspbian_lite-2016-02-29`.

I've downloaded this image and performed the same install and setup to run the Neopixel script. The result is again positive.

Identification of this system:

```bash
$ uname -a
Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ ls -l /boot/kernel.img
-rwxr-xr-x 1 root root 3963140 Feb 26  2016 /boot/kernel.img
```
That seems a match with the original system that worked. So I consider it a positive reference setup.

The system that was failing was based on the currently last release : `	raspbian-2017-04-10   10-Apr-2017 18:51`. This system was fully updated and upgraded.

Maybe this can point someone in the right direction to further debug the difference between these two releases.

SimonJakubowski, 2017-05-12:
Downgrading the Firmware to version 4.9.13 works for me.

`sudo rpi-update 411288a7e31067fffa1b955a30b133bf2902bb54`

penfold42, 2017-05-13:
I wonder if this library is stomping on a DMA channel that was previously unused and is now in use by something.

tateu, 2017-06-03:
I'm having a similar issue with a GRBW SK6812 strip with 204 LEDs.

Raspberry Pi 3B
4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux

The python script always freezes somewhere after 50 or so minutes. In 3 recent tests:

1) Pin 18 (PWM), DMA 5 -- My script froze after 144 minutes
2) Pin 21 (PCM), DMA 5 -- My script froze after 63 minutes
3) Pin 21 (PCM), DMA 10 -- My script froze after 71 minutes

I've also had SK6812_strandtest.py freeze at around the 50 minute mark.

If I kill the python process from another terminal (ctrl C does not work when it is frozen), I can immediately restart the script and it will work again for awhile.

I cannot get SPI (Pin 10) to work at all. My strip just flashes randomly or goes to full brightness.

I'm going to try Firmware version 4.9.13, as suggested by SimonJakubowski.

tateu, 2017-06-15:
I found the problem I was having and it looks like it is different than yours because it was caused by a commit made on May 31 (https://github.com/jgarff/rpi_ws281x/commit/6c0f4db06913fc796743b2c666a60fbe2a287c1a#diff-ad15add6b56882c8702200b9e4d154d8R1194) and a 2^64 overflow back to 0;

```
ws2811->render_wait_until = 18446744073709290267

const uint64_t current_timestamp = get_microsecond_timestamp() = 234440

usleep(ws2811->render_wait_time - time_diff) = usleep(18446744073709055827)
```

I fixed it with https://github.com/tateu/rpi_ws281x/commit/3a0447e5127ea27a3f85ab2122202efd970b37f6. So far, I have been running a python script with PCM for about 6 hours without any issues and it's still going. That is the longest I've ever had it run.

SPI still doesn't work at all for me, though.

shortbloke, 2017-06-16:
@SimonJakubowski Are you still having issues? If you leave it running have you noticed your file system going read only? As with the latest Raspbian updates I see some of the errors you're seeing and also errors from mmc per this issue: https://github.com/jamesbulpin/meshblu-connector-ws2811/issues/4

SimonJakubowski, 2017-06-27:
Have the same error with the latest Raspbian (even with the old Firmware), but by changing the DMA channel to 10 the issue is resolved for me.

l-spitz, 2017-08-04:
Just ran into this issue, and changed channels to 10, which has greatly improved things, but still see this message in syslog

`kernel: [14231.478962]  JBD2: Detected IO errors while flushing file data on mmcblk0p7-8`

It happens each time I call the NeoPixel library.  Is it safe to ignore?

closed 2021-12-03 by Gadgetoid