jgarff/rpi_ws281x #179: strandtest.py

gscott1962, 2017-05-12:
When I run this program the leds are erratic with no pattern stops starts what am I doing wrong?

JFTack, 2017-05-13:
I got some similar problem,
But when I make a script that should show only LED 1 it seems that every time the same LEDs are powered, so multiple LEDs, not no. 1, and different colors...
I have this strip: LS12RGB60WS2811I with a length of 2 meter.
I tried with and without logic level shifter, same result...

penfold42, 2017-05-13:
Have you disabled analog audio ?

JFTack, 2017-05-14:
Well, last night I did read about that, disabling the audio in the configuration.txt works like charm...
It was a short night but they look awesome (H)

gscott1962, 2017-05-14:
How do you disable the audio?

JFTack, 2017-05-15:
If you search around you can find more info about it.
I commented a line in the /boot/configuration.txt. Just put a # In front of this line dtparam=audio=on.
The following are 2 other options, I didn't need to use them...
Check the /etc/modules file.
Add blacklist snd_bcm2835 in /etc/modprobe.d/snd-blacklist.conf
Some seem to get it working by just forcing audio to HDMI, so if you want audio you can try that first.
Add hdmi_force_hotplug=1 and hdmi_force_edid_audio=1 to /boot/config.txt.
Also Force Audio to HDMI through raspi-config.

closed 2017-09-16 by jgarff