jgarff/rpi_ws281x #180: Strandtest.py error

Bailplay, 2017-05-15:
Hello 
I followed the instructions on https://learn.adafruit.com/neopixels-on-raspberry-pi/software but im always getting this Error code when im starting the strandtest.py:
Traceback (most recent call last):
  File "neopixel.py", line 52, in <module>
    class Adafruit_NeoPixel(object):
  File "neopixel.py", line 54, in Adafruit_NeoPixel
    brightness=255, channel=0, strip_type=ws.WS2811_STRIP_RGB):
AttributeError: 'module' object has no attribute 'WS2811_STRIP_RGB'

what can i do?

penfold42, 2017-05-28:
Have you updated your repository from an older version ?

I suspect something isn't magically cleaning up if you update and rebuild.

Bailplay, 2017-05-30:
Im sure that I have updated nothing(Im not an expert so I dont really know if i did). Maybe it could help that i reinstalled several times the operating system ubuntu mate. And i also tryed other instructions but im always getting the same problem and thats very weird.

mitchcoop16, 2017-06-21:
hello
I followed the steps exactly from the adafruit website and I'm getting the same error.
File "/home/pi/rpi_ws281x/python/examples/strandtest.py", line 8, in <module>
        from neopixel import *
ImportError: No module named 'neopixel'

How can I fix this? I'm using the raspberry pi 3 and fairly new with it

skaldo, 2017-07-09:
@mitchcoop16 try to install the module using `python setup.py install` in the `python` directory.

mitchcoop16, 2017-07-09:
@skaldo I've tried that as well. It says there is no such file or directory.

skaldo, 2017-07-09:
@mitchcoop16 
```
cd /home/pi/rpi_ws281x/python
python setup.py install
```
Assuming, that `/home/pi/rpi_ws281x` is the directory where you cloned this repo into.

mitchcoop16, 2017-07-10:
@skaldo That did the trick!! thanks for the help!

pranavgaddamadugu, 2017-07-14:
I'm having the same issue, using Ubuntu Mate. Does the operating system have something to do with it?

Bailplay, 2017-07-14:
@pranavgaddamadugu I changed to Rasbian instead of using Ubuntu and it worked. Try it out maybe it works.

jgarff, 2017-09-16:
Closing due to inactivity.  Please re-open if necessary.  Thanks.

closed 2017-09-16 by jgarff

whashburn, 2017-11-22:
Thank you for your libraries and support.  

I suspect the error people have been experiencing, myself included, is because we are trying to run under Python 3 and not Python 2.   If you have time would you be able to update the installation instructions for us 'bilingual' <grin> python users?

When I loaded the example in Python 2 it now fails with at line 102: strip.begin() with a runtime errorws281_init failed with code -5 (mmap() failed).  I'll keep plugging away.

jgarff, 2017-11-22:
> 'bilingual' python users?

Generally mmap() failures are due to permissions or a kernel configuration
that doesn't include Broadcom video support.  I don't know that this is a
python issue.  Make sure you're running as root and using a supported
kernel.

whashburn, 2017-11-23:
Thank you for your quick reply Jeremy. Good news and disappointing news.
I removed the rpi_ws281x folder and re-installed.  When I run strandtest,py into Python 2 IDLE I still receive the ws2811_init failed with code -5 (mmap() failed).

However, - good news, if I run the command as described in readme.md strandtest.py runs correctly.

I've need to do some debugging of my config.  Pi2 running raspbian-stretch.

If you are in the US - Happy Thanksgiving.  If you are anywhere else - Happy Thursday  :-)

Thanks again

whashburn, 2017-11-23:
Hi Again, For the benefit of someone that might come across this thread - running IDLE (2 or 3) as root is risky.  An safer solution I came across is  (A) open your python files in IDLE to edit, and (B) open another terminal window on the desktop, where you may type sudo python yourscripy.py and have it run with escalated privileges

hlc111, 2017-12-17:
For Ubuntu Mate swig default version is 3.0.8. Install swig v3.0.10 from source code  fixed this issue.