jgarff/rpi_ws281x #182: Flickering when using same power source

salehtahini, 2017-05-19:
I followed this tutorial https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring using the 74AHCT125 IC and used two power supplies, 2A charger for the Pi Zero and a 6A for the strip. After I ran the test script and found that everything is working fine, I tried powering the Pi and the strip from the same 6A power supply. I didn't change anything with the wiring.
when I used the same power supply the LEDs started flickering and showing random colours, I tried cutting the strip leaving only 4 LEDs connected to the Pi and it's still flickering, so it's definitely not an ampere problem.
what should I do? I need to power the strip and the Pi from the same source for my project and this library isn't working when I use the same power supply.

penfold42, 2017-05-19:
Did you use hdmi the first time and not use hdmi the second time ?

If so, you need to disable analog audio

salehtahini, 2017-05-19:
Using HDMI in both runs, and the analog audio is disabled and blacklisted.

dstaley, 2017-05-25:
I'm running into the same issue, except when the Raspberry Pi and LEDs are run from the same power source, the LEDs don't functional at all. Simply moving the Raspberry Pi to its own power source corrects the issue.

Since this is a power-related issue, I doubt it's the libraries fault, but I just wanted to chime in just in case it was!

penfold42, 2017-05-25:
Got a picture or diagram showing exactly how you have wired it up ?

I have a test board that I use with Hyperion and it has 2 leds - one pwm and one spi attached.

It has a 74HCT based level shifter on it - maybe you should both look into a level shifter?

In theory it's required - in practice it will depend on lots of things.

dstaley, 2017-05-25:
@penfold42 Here's a rough diagram of my circuit. I apologize in advance as I'm a total beginner when it comes to Fritzing.

[image]

Essentially the 5V power supply terminated in a screw terminal, which I have a capacitor and four wires plugged into (two power lines and two ground). Two wires go to the first Neopixel ring, and the other two go to a microUSB connector to power the Raspberry Pi.

I don't have a level shifter attached, but I'll see if I can order one and give that a try.

dstaley, 2017-05-27:
Just wanted to confirm that after adding in a TXB0104 level shifter that I'm not longer experiencing any issues powering the LEDs and the Raspberry Pi from the same power supply.

closed 2017-09-16 by jgarff