jgarff/rpi_ws281x #184: RPI 2B+ and SPI

vgrillot, 2017-05-25:
Hi, 
First thanks for you port to python for this lib !
I'm successfully using it with PCM gpio 18 output. I use to run a show on a led strip, used in a pinball.

As I want to use also the analog sound output, I'm trying to switch to SPI output, pin 10 (MOSI).
I'm using a RPI 2B+, python3.4, the spi driver is enable (using raspi_config) and I can see /dev/spidev0.0 and spidev0.1. I'm using sudo to run script to get full access, just in case.

Unfortunately I can only got the same error.

```
Traceback (most recent call last):
  File "python/examples/pinball.py", line 187, in <module>
    strip.begin()
  File "build/bdist.linux-armv6l/egg/neopixel.py", line 115, in begin
RuntimeError: ws2811_init failed with code -7 (Unable to initialize GPIO)

```

Here the config I'm using:

```
# LED strip configuration:
LED_COUNT      = 64      # Number of LED pixels.
# GPIO pin connected to the pixels (must support PWM!).
#LED_PIN        = 18      # PWM (ok)
LED_PIN        = 10      # MISO (SPI)
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 1       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)

```

It's probably not an issue but I'm trying to send help request as I'm not sure where to ask !
My script is a derivated from strandtest.py

Should it works "plug'n play' just chaning the LED_PIN ?
Would you have any python sample script to run with the SPI output ?

Thanks for support,
Vincent.

AlessandroFerri, 2017-05-26:
Hi,
I also have a similar problem. I am using a raspberry Pi3. I've done tests with PWM0, PWM1, PCM_DOUT, either by directly connecting the LED Din port to the respective port of the raspberry GPIO, both using the Level-converter Chip wiring 74AHCT125. It's all a miracle. But I need to use SPI0-MOSI. I did a test by connecting the LED Din port directly to the GPIO10 (MOSI) of raspberry, I made the appropriate changes to main.c:
`#define GPIO_PIN 10`
and on strandtest.py:
`LED_PIN = 10`
The programs run without any error, but the LEDs are turned on completely white and the raspberry crashes.
Maybe I made a wiring error?
Can anyone tell me how do I wiring either directly to the led or using the 74AHCT125?

Thanks for your help.

penfold42, 2017-05-28:
@vgrillot 

I've just been updating the documentation and playing with strandtest.py
The only change I've made is to the LED_PIN and it works fine for me with a Pi3

Did you have an older version of this installed previously? Maybe the python path is finding that first

AlessandroFerri, 2017-05-29:
Hi vgrillot,
I installed jgarff/rpi_ws281x from master one week ago. Is there a Python version I have to install? Maybe I did not wiring well the gpio. You can tell me how wiring the raspberri gpio with the led as directed as with 74AHCT125 for SPI0-MOSI

Thanks for your help.

vgrillot, 2017-06-11:
Hi @AlessandroFerri,
please follow this link, it helps me a lot :  [http://dordnung.de/raspberrypi-ledstrip/ws2812](url)
it's a very well done tutorial.

@penfold42,
I continue to struggle a bit,
It works fine if I use the C example.
I finally discovered that it works with python3.4, and not python2.7.
I'm not sure what's wrong, but as my main project is running under python3, i'm fine for now.

By the way, is the pip package "rpi_ws281x" updated ?
I tryed it with errors, and if I set the python path to my cloned dir it works.

Thanks for all,
Vincent.

marbaque, 2017-11-07:
@penfold42 what PIN do you use instead of 18?

Gadgetoid, 2018-03-06:
@vgrillot the pip package rpi_ws281x is generally up to date, but is currently built from my python bindings found here: https://github.com/pimoroni/rpi_ws281x-python

It works in both Python 2 and 3 and forms the back-end for Pimoroni's Unicorn HAT library.

vgrillot, 2018-03-06:
Hi guys,
I forgot to answer, but yes I finally ran it successfully after a new git
or package update.
Thanks again !
Vincent.

closed 2018-03-06 by Gadgetoid