jgarff/rpi_ws281x #185: Color( green, red, blue)?

azeunkn0wn, 2017-05-26:
Even in strandtest.py, the colorwipe starts at color green, then red, when the code says it should start with red then green.

```
while True:
		# Color wipe animations.
		colorWipe(strip, Color(255, 0, 0))  # Red wipe
		colorWipe(strip, Color(0, 255, 0))  # Blue wipe
		colorWipe(strip, Color(0, 0, 255))  # Green wipe
```

penfold42, 2017-05-26:
The default strandtest probably assumes a ws2811 based led strip with the colour order of red, green, blue.

Wa2812 leds with the integrated controller expect a colour order of green, red, blue.

The https://github.com/jgarff/rpi_ws281x/blob/master/python/examples/SK6812_strandtest.py example shows how to set the strip type.

The strip types are defined in https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.h

azeunkn0wn, 2017-05-26:
Oh! I get it now.

closed 2017-05-26 by azeunkn0wn