jgarff/rpi_ws281x #186: MFRC522 issue

azeunkn0wn, 2017-05-26:
I use 2x RC522 (RFID Reader), via SPI. (CE0 and CE1 for SDA, GPIO24 and 25 as rst, MOSI,MISO, and SCK)
using pirc522 module

I have 270 LED strip (will only use 29 of them), connected to my PC's power supply (GND and +5V), and the Data In and GND is connected to RPi's pin 18 and GND respectively.

the issue is, when the python script with rpi_ws281x module started, my RC522 became unstable. 
I have regular 3V led connected (+3.3v >> resistor >> LED >> SDA === CE0/CE1) to monitor the interface (it is ok, right?)
by unstable, I mean RC522 will not respond for a few seconds, or LED indicators on SDA flickers a lot.

am I doing something wrong? or am I missing something?

```
# LED strip configuration:
LED_COUNT      = 29     # Number of LED pixels.
LED_PIN        = 18     # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor)
```

closed 2017-05-26 by azeunkn0wn

penfold42, 2017-05-27:
This library is hardxodes to use spidev0.0 which probably uses CE0 as the slave select.

The leds don't care about slave select so any spidev0.x spi access will likely drive the leds nuts

azeunkn0wn, 2017-05-27:
That 'issue' was just probably a false alarm. my later tests was ok, and running perfectly, ignoring the SDA LED indicators since it is probably normal.
I'm not getting that long pause after detection and unresponsiveness in my RFID. Though, I had to test some more.

Is my setup ok? or do I have do some minor changes (with the LED strip config) to have smooth operation?
I like to hear some advice, if you don't mind.