jgarff/rpi_ws281x #188: SD Card Errors after using Library

encelius, 2017-05-28:
When running the ./test file, the LEDs behind the rainbow are not shut off as I think they are supposed to (Problem can be seen in this video: https://cloud.kruben.de/index.php/s/kDlOQnOcaB4EWeS).
Also, whenever using the library, be it via the test file, strandtest.py or otherwise, dmesg shows errors with the sd card. Here are the last few lines of dmesg after running ./test, complete output of dmesg is in the attached .txt file:

[  323.759406] mmc0: =========== REGISTER DUMP ===========
[  323.772459] mmc0: SDCMD  0x00000052
[  323.785777] mmc0: SDARG  0x010d0910
[  323.799076] mmc0: SDTOUT 0x017d7840
[  323.812278] mmc0: SDCDIV 0x00000006
[  323.825548] mmc0: SDRSP0 0x00000900
[  323.838695] mmc0: SDRSP1 0x00001269
[  323.851548] mmc0: SDRSP2 0x7fffffff
[  323.864175] mmc0: SDRSP3 0x00024007
[  323.876795] mmc0: SDHSTS 0x00000001
[  323.889367] mmc0: SDVDD  0x00000001
[  323.901659] mmc0: SDEDM  0x00010902
[  323.913853] mmc0: SDHCFG 0x0000040e
[  323.925940] mmc0: SDHBCT 0x000001c0
[  323.937854] mmc0: SDHBLC 0x00000010
[  323.949841] mmc0: ===========================================
[  323.962544] mmcblk0: error -110 transferring data, sector 17631504, nr 16, cmd response 0x900, card status 0xb00
[  323.967473] mmcblk0: retrying using single block read
[dmesg.txt](https://github.com/jgarff/rpi_ws281x/files/1034454/dmesg.txt)

penfold42, 2017-05-28:
I found on my OSMC system that it interfered with SPI transfers.

In my case changing the DMA channel to 10 seemed to fix it.

encelius, 2017-05-29:
Unfortunately this didnt fix the problem for me. Are there any other DMA channels i should try or a resource where i could find out more about DMA channels?

penfold42, 2017-05-29:
Can you use gpio 10 ? This uses spi instead of low level hardware stuff like pwm it pcm do.

It also doesn't use dma directly - the kernel spi driver does it cleanly.

penfold42, 2017-05-29:
What OS are you running ?

encelius, 2017-05-29:
I will try this when I come home later. I'm running Arch Linux on my pi

encelius, 2017-05-29:
It didnt work with gpio 10 so far either. With DMA channel 10 I got a lot of flickering and with DMA channel 5 it just lights up in seemingly random colors. But at least this time I didnt get any error messages in dmesg.

shortbloke, 2017-06-20:
I think this is the same issue I'm seeing. If you leave the program running I found it eventually hits enough errors to mark the file system read only. See: https://github.com/beyondscreen/node-rpi-ws281x-native/issues/61

encelius, 2017-06-20:
Indeed, this seems to be the same problem. So far I didn't find any root causes though.

skakri, 2017-06-20:
@encelius which Pi are you using?

encelius, 2017-06-21:
I'm using a raspberry pi 3 model b

shortbloke, 2017-07-06:
I've just tested using DMA 10 via the command: 
```
./test -x 144 -y 1 -c -d 10
```
This works fine, if I remove the the `-d 10` then it once again breaks. I'm not changing anything else in my setup and haven't had flicking issues.

encelius, 2017-07-06:
I just pulled and built anew, but unfortunately it still doesnt work. I tested it with
`sudo ./test -x 100 -y 1 -c -d 10`
and both ended up with flicking LEDs. Also, it didnt shut the LEDs down afterwards.

shortbloke, 2017-07-06:
@encelius But you don't get flickering on DMA 5? 
I had flickering issues at various stages of playing with my WS281x 144 led strip. I had to address it with 74hct125 to shift the data signal from 3V to 5V, but then also have this output connect to the LED via a 470ohm resistor.  However that was when using a different PSU for Pi and LED strip, when using a the same 5V 10A PSU for both I can acutally skip the level shifter and get the data line directly from the Pi GPIO.

I suspect the flickering is a different issue to the DMA conflict and memory card errors.

encelius, 2017-07-06:
I do have a level shifter as well, but Im not using it at the moment as i initially thought the problem would stem from the fact that my pi is putting out more power than it can handle with powering the level shifter via its 5V pin, and ive not yet put it back in (I cant power the level shifter with my power supply since its 12V, as my LED strip needs 12V but has a 5V data input). With
`sudo ./test -d 5 -c`
the test sequence proceeds over the strip seemingly normally, but all other LEDs are flickering in green light all the while. Also, at the end it shut off all LEDs but the last

shortbloke, 2017-07-06:
Can you point me at the spec for your LED strip? Often times the data line input voltage will have a relationship with the overall LED strip voltage, so perhaps for your 12V LED strip the data input voltage needs to be higher. 
Also ensure since you are using 2 PSU that you have connect ground/negative from your Pi GPIO Header and from your 12V PSU so that they share a common ground.

shortbloke, 2017-07-06:
Actually from a quick bit of reading the 12V LEDs use the same 5V Signalling. So I'd ensure you have a common ground, if so then try adding back your level shifter to see if that helps get the 3v3 Pi data output up to a higher voltage.

shortbloke, 2017-07-06:
Also ensure you have disabled audio (i had this as a cause of flickering). This issue is in the known issues of the readme IIRC. However, the method I used was to edit the boot/config.txt and comment out the line as per below:
```
# Enable audio (loads snd_bcm2835)
#dtparam=audio=on
```

encelius, 2017-07-06:
The strip is the one i posted in my original post.
If PSU means powersupply, then you're partly correct, I'm using only one and I'm powering the level shifter with the 5V pin of the pi (pin 02).
I added the level shifter, now it's brighter, but still only flickering. Also, when i just tried using the DMA 10 again (the default one) this time i got my filesystem to remount readonly pretty much instantly.
I already blacklisted the audio driver snd_bcm2835 via a blacklist .conf file in /etc/modprobe.d and ensured via lsmod that its not loaded.

shortbloke, 2017-07-06:
Ok so you have Pin 02 (5v) powering the level shifter, you've got pin 12 (GPIO_GEN1) as the data input to your level shifter, and you've got the Pi Ground Pins 06, 14, 20, 30, 25, 09 as the ground for your level shifter. Correct so far?
Have you also connected the ground from your 12V supply to the ground of your level shifter? 
Pretty much like the diagrams here: https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring with the exception that your input is 12V, so you need to source 5V for the shifter from the Pi and avoid conencting the 5V to the 12V.

encelius, 2017-07-06:
Almost, yes. I've got all ground on pin 06 and I imagine my level shifter is a little bit different than the one in your diagram, I've used this datasheet for connecting: http://www.ti.com/lit/ds/symlink/sn74ahct125.pdf where i used GND as ground pin, VCC as power source connected to the 5V output of the pi (pin 02), 1A as data input (GPIO18) and 1Y as data output going into the LED strip.

shortbloke, 2017-07-06:
No you've got the same level shifter as shown on the adafruit page. Have you connected pin 1 on your sn74ahct125 (1OE) to ground? It's not sufficient just to have 1A, 1Y, VCC and GND Connected you also need 1OE connected to ground, it can't be left floating.

encelius, 2017-07-06:
Ah, I didnt know. Thanks for the tip! Sadly it didnt really resolve the problem, though now the -c flag to switch off the LEDs after killing the test program seems to work.

shortbloke, 2017-07-07:
I know my setup is 5V, so easier as I can use the same PSU to power the Pi, the level shifter and the light strip. I've found that when using the same PSU for Pi and strip that it actually works well without the level shifter, but when using 2 different PSU I needed it. Anyway this is what I have working on my bench right now: 
[image]

(not shown, but for the cable joins I'm using wago connectors to make it easier to join the mix of stranded wire and solid core patch leads I'm using).

Finally it would appear the original issue is resolved by moving to DMA 10 rather than 5. Other issues with flickering LED probably ought to be in a different issue/ticket.

electrokean, 2017-07-07:
When wiring up a 74AHCT125, make sure all inputs are not left floating or the chip may draw excessive current.  Connect them to anything GND, signal, whatever - just don't leave them floating.

encelius, 2017-07-07:
Alright, just downloaded Fritzing to make a sketch myself to be able to show it - hope its readable enough. Didnt find all parts that i needed, the level shifter I'm actually using is a SN74AHCT125N if thats relevant. Also, when i get back home I'll try and connect the rest of the pins of the level shifter to ground just to make sure that thats not the problem, thanks for the tip.

[image]

encelius, 2017-07-07:
Alright, it finally seems to work after I connected all the inputs of the level shifter I'm not using to ground, thanks electrokean for the tip! And thanks shortbloke for the tip with the DMA 10! So far it worked flawlessly with
`sudo ./test -d 10 -c`
and
`sudo python strandtest.py`
after configuring strandtest.py to use DMA 10 instead of 5.
Big thanks, I appreciate the help!

closed 2017-07-07 by encelius