jgarff/rpi_ws281x #190: Errormessage after executing strandtest.py

LK608, 2017-05-29:
Hi,
I'm using Ubuntu MATE on my Raspberry Pi
After installing the library on the Pi and trying to execute the strandtest.py file I get this error:
`Traceback (most recent call last):`
  `File "strandtest.py", line 8, in <module>`
    `from neopixel import *`
 ` File "build/bdist.linux-armv7l/egg/neopixel.py", line 52, in <module>`
 ` File "build/bdist.linux-armv7l/egg/neopixel.py", line 54, in Adafruit_NeoPixel`
`AttributeError: 'module' object has no attribute 'WS2811_STRIP_RGB'`

Does somebody know what I did wrong?
Thanks in advance
LK608

penfold42, 2017-05-29:
Did you recently update your got repository and rebuild ?

If so, try removing the build folder and rerunnning the python build.

makelove, 2017-11-15:
Ubuntu MATE 16.04 on Raspberry Pi 3b
```
pi@pi-desktop:~/github/rpi_ws281x/python/examples$ python3 strandtest.py
Traceback (most recent call last):
  File "strandtest.py", line 8, in <module>
    from neopixel import *
  File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x-1.0.0-py3.5-linux-armv7l.egg/neopixel.py", line 52, in <module>
    class Adafruit_NeoPixel(object):
  File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x-1.0.0-py3.5-linux-armv7l.egg/neopixel.py", line 54, in Adafruit_NeoPixel
    brightness=255, channel=0, strip_type=ws.WS2811_STRIP_RGB):
AttributeError: module '_rpi_ws281x' has no attribute 'WS2811_STRIP_RGB'
```

Have you fixed this problem yet ?
Is because the OS?must be on Raspbian ?

-
I am following http://frederickvandenbosch.be/?p=1014 tutorial

Evader611, 2017-11-22:
I'm having the same issue.  Tried deleting build and re-running sudo python setup.py install.  Then cd to examples and sudo python strandtest.py results in the same output as listed by @makelove

Evader611, 2017-11-22:
This appears to be a Python Wrapper problem only.  In the ~/rpi_ws281x directory, I can run sudo ./test successfully.  Really prefer to do all of this in Python, not c

Frodox, 2017-12-03:
@LK608 which is exact commands did you use?
I have no problem neither with python2 nor python3 when following README.md.
Could you do clean test and provide exact commands? which python version is default on mate, 2/3?

LK608, 2017-12-04:
Hey there,
It's now working for me!
Thanks for all the help.

closed 2017-12-04 by LK608