jgarff/rpi_ws281x #193: Error building on RPi 2 model B

mike-strauch, 2017-06-05:
When running `scons`on RPi 2 model B, I receive the following error:

`rpi_ws281x/ws2811.c:126 undefined reference to clock_gettime`

I've seen some suggestions on stack overflow to upgrade glibc to a newer version to fix it, but I'm curious to know if you have any other suggestions before I do that.  

Thanks!

penfold42, 2017-06-05:
Is that the only abnormal output from scons ?

What OS is on your Pi ?

sschroe, 2017-06-05:
In `SConstruct` please try adding -lrt to the `LINKFLAGS`. This should look like this:

```
platforms = [ 
    [
        'userspace',                # Target Name
        'linux',                # Scons tool (linux, avr, etc.)
        {                       # Special environment setup
            'CPPPATH' : [
            ],
            'LINKFLAGS' : [
                "-lrt"
            ],
        },
    ], 
]
```

With LINKFLAGS initially being empty.

mike-strauch, 2017-06-06:
> In SConstruct please try adding -lrt to the LINKFLAGS.

That seems to have done the trick, thanks!

> What OS is on your Pi ?

It is Raspbian 7 wheezy.

sschroe, 2017-06-06:
I'll look into adding the link flag generally to support the older dist's. With Raspbian Jessie it works without it due to a newer glibc i suppose.

mike-strauch, 2017-06-25:
I'm having a similar issue when attempting to run the python examples inside `/python` (namely strandtest.py).  I've followed the instructions in the `README.md` to install swig and run the setup script; however when running `sudo PYTHONPATH=".build/lib.linux0armv7l-2.7" python examples/strandtest.py`, I get basically the same error as before:

```
ImportError: /root/.python-eggs/rpi_ws281x-1.0.0-py2.7-linux-arm7l.egg-tmp/_rpi_ws281x.so: undefined symbol: clock_gettime
```

I can provide the full stack if you'd like, I'm just having some issues with wifi on my pi at the moment.

penfold42, 2017-06-25:
You've got a type on your PYTHONPATH.

It's missing a colon

mike-strauch, 2017-06-27:
> You've got a type on your PYTHONPATH.
It's missing a colon

Apologies, that is indeed a typo, but it's from me typing the command by hand rather than copy/paste into my comment.  I'm using the exact command contained [here](https://github.com/jgarff/rpi_ws281x/blob/master/python/README.md#run-a-demo).

mike-strauch, 2017-07-05:
Does the python wrapper build use a local copy of the C implementation of the ws281x library or is it pulled from github or some other remote end point?  I'm not too familiar with the build tools being used for the python portion of the project.  It just seems like maybe it's not using my locally built copy of the C library to build the python wrapper.

PR #195 (2017-07-06): Add -lrt link flag for systems with older glibc
This adds -lrt to the link flags to be able to link on systems with older glibc versions where -lrt is required for `clock_gettime`. For newer glibc versions it has no influence.

Closes #193.
closed 2017-07-06 by jgarff