jgarff/rpi_ws281x #194: SPI frequency changes when cpu is throttled

hungerpirat, 2017-06-06:
I have been playing around with the SPI mode on a RPi3. 
my config: 

# LED strip configuration:
LED_COUNT      = 7      # Number of LED pixels.
# GPIO pin connected to the pixels (must support PWM!).
LED_PIN        = 10      # MISO (SPI)
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)

Running the test script:
In the oscilloscope I see a proper signal, but the frequency is at 500kHz, all LEDs are white. 
If I put some load on the RPi3 cpu, it switches to the specified 800kHz and I see the rainbow like expected. 
Once the cpu load has been gone, it switches back to 500kHz and all LEDs are white again.

Maybe this bug is related to #184

hungerpirat, 2017-06-06:
Just checked the cpu frequency while running the test. The phenomenon is clearly related to the cpu throttling state.

hungerpirat, 2017-06-06:
more test results: 

SettingLED_FREQ_HZ to 1200000 works in both cpu speeds 600MHz and 1.2GHz. The resulting SPI frequencies are 600kHz, respectively 1.2MHz. 
The data sheet of ws281x LEDs mentions bit timings of 1.25µs with ±600ns tolerance. The calculated valid freq range is 540kHz to 1.5MHz and includes both of the frequencies above.

penfold42, 2017-06-06:
Strictly speaking it's the core clock speed changes that cause this.

The SPI hardware is set by a divisor and the kernel uses the core clock speed when the device is opened to calculate this.

The clock governors then mess with this to save power and temperature which inadvertently impacts the spi clock.

The best solution is for the kernel to also recalculate the divisor.

One workaround is to force_turbo=1

hungerpirat, 2017-06-07:
Sure - I was wondering if there is any chance to get the core clock change
as an event and adjust the spi speed accordingly. I think the "all white"
phenomenon of #184 is related to switching down from full speed to
throttled. If the SPI clock is set to the standard value of 800kHz the
down-throttled value is that slow that every sent symbol is interpreted as
1 - which results in all white on the strip.

penfold42, 2017-06-07:
The kernel should be doing it - user space asks for a speed and at the moment, this isn't guaranteed.

Something to investigate is repeatedly opening and closing the device on every write. 
Assuming there's no great overhead, this should work better.
If changes happen I'd write, it won't help for that update tho.

penfold42, 2017-06-07:
Btw,
https://github.com/raspberrypi/linux/issues/2060#issuecomment-306756533

hungerpirat, 2017-06-07:
thanks - that's very helpful.
If I will find some time, I will test your proposal about opening the device every time.

penfold42, 2017-06-07:
I don't think it works - I tried it with Hyperion.

Re-reading the responses to the other issue implies that the spi subsystem is lied to with the highest possible core clock frequency - it doesn't appear to be a dynamic.

wheresthecode, 2017-07-09:
I was running off PWM but am trying to switch to SPI to free up PWM for audio. I added core_freq=250 to /boot/config.txt which seems to work. However, I've noticed that all the colors are incorrect.

I think the signal is getting shifted because if I try to set the LEDs to red, I will get all yellow LEDs except the first LED will be white.

I'm on a pi3, python 2.7. LED frequency is set to 800000. Coloring order is WS2811_STRIP_GRB which worked as expected using PWM and GPIO pin 18.

Any ideas how to correct this color shift?

timvandenhof, 2019-12-23:
> I was running off PWM but am trying to switch to SPI to free up PWM for audio. I added core_freq=250 to /boot/config.txt which seems to work. However, I've noticed that all the colors are [...]

It's a while ago now, but have you found a way to resolve your issue? I might be facing something similar. I opened a tread about it on the Raspberry Pi Forums: https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=260037

The issue I'm facing is that it works fine over PWM, but using SPI results in wrong white flashing colors, where one LED lights up green. I've seen issues with python2 and python3, but also experienced issues with .Net Core on Raspbian https://github.com/dotnet/iot/issues/257

Still investigating, I'm expecting a raspberry pi configuration error / user mistake, not a library issue. Hence the tread over at the Rpi Forum.

stephbellemare, 2024-09-22:
> > I was running off PWM but am trying to switch to SPI to free up PWM for audio. I added core_freq=250 to /boot/config.txt which seems to work. However, I've noticed that all the colors are incorrect.
> > I think the signal is getting shifted because if I try to set the LEDs to red, I will get all yellow LEDs except the first LED will be white.
> > I'm on a pi3, python 2.7. LED frequency is set to 800000. Coloring order is WS2811_STRIP_GRB which worked as expected using PWM and GPIO pin 18.
> > Any ideas how to correct this color shift?
> 
> It's a while ago now, but have you found a way to resolve your issue? I might be facing something similar. I opened a tread about it on the Raspberry Pi Forums: https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=260037
> 
> The issue I'm facing is that it works fine over PWM, but using SPI results in wrong white flashing colors, where one LED lights up green. I've seen issues with python2 and python3, but also experienced issues with .Net Core on Raspbian [dotnet/iot#257](https://github.com/dotnet/iot/issues/257)
> 
> Still investigating, I'm expecting a raspberry pi configuration error / user mistake, not a library issue. Hence the tread over at the Rpi Forum.

use this:
On an RPi 3 you have to change the GPU core frequency to 250 MHz, otherwise
the SPI clock has the wrong frequency.

Do this by adding the following line to /boot/config.txt and reboot:

```
    core_freq=250
```

On an RPi 4 you must set a fixed frequency to avoid the idle CPU scaling changing the SPI frequency and breaking the ws281x timings:

Do this by adding the following lines to /boot/config.txt and reboot:

```
    core_freq=500
    core_freq_min=500
```

SPI requires you to be in the `gpio` group if you wish to control your LEDs
without root.