jgarff/rpi_ws281x #196: strip.show() seems to hang after a long time

Anfield800, 2017-06-17:
I am a beginner so hopefully I've just made a simple mistake that someone will spot quickly.

I wrote a small piece of python code to show hours and mins on a 72 LED strip.  The code works well initially but then stops.  I've done what debugging I can and the problem seems to be when using the strip.show() function.  For some reason it, after a variable amount of time, it takes ages to return, sometimes over an hour.

I've checked my code, so as a test I ran the example strandtest.py and left it running. This did exactly the same thing after about 45 mins or so.

I'm using a level converter chip, 74AHCT125 on the data line to the strip which I have swapped just in case the first one is faulty. 

My wiring is as per https://learn.adafruit.com/neopixels-on ... -pi/wiring however my Pi is powered separately from the LED strip. I have added a capacitor and resistor as per this advice -https://learn.adafruit.com/adafruit-neopixel-uberguide/best-practices but this hasn't made any difference.

Would anyone have any ideas?

tateu, 2017-06-17:
I had the same issue and solved it by modifying code in rpi_ws281x. I have now successfully run a script for over 48 hours. See my two comments here: https://github.com/jgarff/rpi_ws281x/issues/176#issuecomment-305954369. Try my fork which has fixed that issue for me, https://github.com/tateu/rpi_ws281x, and let me know if it works for you.

PaddyderBOY, 2017-06-18:
Thanks alot, my code is running more than 30 min now :)

Anfield800, 2017-06-18:
Thanks very much, don't know why I didnt spot this when I was searching for answers. I've changed the code and am running a test now.

Anfield800, 2017-06-18:
So unfortunately that didn't work for me. Still having the same problems. I have found a work round. As I only change the LEDs every 60 seconds, I call strip.begin() before every change at it works.

tateu, 2017-06-18:
How did you apply the changes? Did you download my fork into a new folder or did you make the changes in an existing folder?

You might want to make sure that the changes were really installed for Python. When I first started making changes to the C code, the changes were not being compiled into the Python lib because I was using an existing folder and no changes were made on the Python side so it was not being recompiled.

To make sure the Python side is recompiled, my full build commands are

```
scons
cd python
rm -rf build
rm -rf dist
rm -rf rpi_ws281x.egg-info
python setup.py install --record files.txt
```

and I make sure an updated version is installed in /usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg
/usr/local/lib/python2.7/dist-packages/rpi_ws281x-1.0.0-py2.7-linux-armv7l.egg

Anfield800, 2017-06-19:
Thanks, I'lll give it another go.

Anfield800, 2017-06-19:
Josh,

That looks like its fixed it.  Been running a couple of hours now with no problems.  Thanks so much, I'm really grateful.

Lee

penfold42, 2017-06-19:
I just created #197 to get this merged

jgarff, 2017-09-16:
Should be fixed in mainline, please re-open if necessary.  Thanks.

closed 2017-09-16 by jgarff