jgarff/rpi_ws281x #198: Colors not displaying properly

bmcginn1, 2017-06-20:
I have been having issues trying to control a single NeoPixel Diffused 8mm Through Hole LED from Adafruit, using a RPi 3 via SPI on pin 10. On their page it says "WS2812B or SK6812-based" and that they are RGB. 

When I put the following RGB values  I get the following colours. Any ideas on whats up? I changed the strip from the default GRB to several of the RGB strips and I had no success. 

(0, 255, 0) ---> Light Blue
(0,255,0) ---> yellow green
(0,0,255) ---> Blue
(90, 165, 0) --- > Green
(12,243, 0) ---> Red

Gadgetoid, 2018-03-06:
Have you had any luck fixing this issue?

Did you actually manage to generate two distinct colours from the same RGB triplet? Or is that a typo?

Slavkata, 2018-04-22:
I have the same issue. And i did generate different colors from the same triplet. Also the first diode stays lit when i do the "clear on exit" form the python example.

Gadgetoid, 2021-12-03:
Closing since this is stale and I like the number of issues to go down sometimes :laughing: instead of just up :sob: 

Feel free to re-open if it's still relevant.

closed 2021-12-03 by Gadgetoid