jgarff/rpi_ws281x #200: RGBW Neopixel - Wrong colours outputted

darthf1, 2017-06-23:
Hi. Thanks for this great library.

I've purchased a [RGBW Neopixel strip](https://www.adafruit.com/product/2847) and hooked it up to my RPI3. 

After installation, im running the `SK6812_strandtest.py` with the following config:

```
# LED strip configuration:
LED_COUNT      = 144      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0
LED_STRIP      = ws.SK6812_STRIP_RGBW	
```

During the strandtest, i noticed the colors are outputted wrong. The strandtest outputs the colors in the following sequence:
```
		colorWipe(strip, Color(255, 0, 0))  # Red wipe
		colorWipe(strip, Color(0, 255, 0))  # Blue wipe
		colorWipe(strip, Color(0, 0, 255))  # Green wipe
		colorWipe(strip, Color(0, 0, 0, 255))  # White wipe
```

However, im getting the following sequence:
```
Green
Red
Blue
White
```

Is this due to a configuration issue? Or is something going wrong with the library?

penfold42, 2017-06-23:
Try changing the strip type to
 ws.SK6812W_STRIP

darthf1, 2017-06-23:
Nice! Now its going as it should. How could i have known to change the type? I was under the impression i had an SK6812RGBW strip but its defined as a SK6812W?

penfold42, 2017-06-23:
The RGBW specifies the colour order.

Those leds actually require data to be sent in the order of GRBW

Ws2811.h has:
#define SK6812W_STRIP   SK6812_STRIP_GRBW

closed 2017-06-26 by darthf1