jgarff/rpi_ws281x #202: Add effect: Fade

darthf1, 2017-06-26:
Hi, I would really love to use an effect to fade from color1 to color2, and even better would be to face from color1 with brightness1, to color2 with brightness2 in X number of steps. 

How could i best achieve this effect? It could look very much like [this](https://learn.adafruit.com/multi-tasking-the-arduino-part-3/fader)

tateu, 2017-06-26:
In Python, how about:

```
import time
from neopixel import *

# LED strip configuration:
LED_COUNT      = 60      # Number of LED pixels.
LED_PIN        = 18      # GPIO pin connected to the pixels (must support PWM!).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0
LED_STRIP = ws.SK6812_STRIP_GRBW

def fade(r1,g1,b1,w1, r2,g2,b2,w2, steps,interval):
	lastUpdate = time.time() - interval

	for i in range(1, steps + 1):
		r = ((r1 * (steps - i)) + (r2 * i)) / steps
		g = ((g1 * (steps - i)) + (g2 * i)) / steps
		b = ((b1 * (steps - i)) + (b2 * i)) / steps
		w = ((w1 * (steps - i)) + (w2 * i)) / steps

		while ((time.time() - lastUpdate) < interval):
			pass

		# print("{: 3d} ({:0.3f}s): {:03d}, {:03d}, {:03d}, {:03d}".format(i, time.time() - lastUpdate, r, g, b, w))
		color = Color(r, g, b, w)
		for i in range(strip.numPixels()):
			strip.setPixelColor(i, color)
		strip.show()

		lastUpdate = time.time()

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL, LED_STRIP)
strip.begin()

fade(255,0,0,0, 10,255,0,0, 40,0.025)
```

darthf1, 2017-06-27:
Wow this looks great! Im going to try tonight, ill let you know the outcome.

jgarff, 2017-09-16:
Closing issue due to inactivity.

closed 2017-09-16 by jgarff