jgarff/rpi_ws281x #203: LED flickering

dushkostanoeski, 2017-06-27:
I'm using a 130 LED WS2813 strip. When I run the strand test, the first 50ish LEDs are lit up OK, but going further down the strip they start to flicker. I won't be using all 130 LEDs at all time so I tried walking a set of 30 LEDs down the strip and again, they start to flicker once they pass a third of the strip.
 
Is it possible that because I'm using Raspberry Pi Zero that the flickering is happening? 

Also I'm not using a level shifter, the strip is connected directly to the Pi, should I try with one?

shortbloke, 2017-06-27:
I assume you are using a secondary PSU to power the LED strip and not using the headers from the PiZero for power, just signal.
I used the implmentation here: https://learn.adafruit.com/neopixels-on-raspberry-pi/wiring with the 74HCT125. Depending on the PSU I use I found in some cases I had to add a 470 ohm resistor to the output of the level shifter. Giving me stable LEDs over my entire strip.

dushkostanoeski, 2017-06-27:
I probably was editing my comment while you were posting yours :) 

I'm using a mobile charger as PSU, because 5Vs are enough for the rPi and the LEDs.

shortbloke, 2017-06-27:
The voltage isn't the issue is the amount of current the PSU can supply. See: https://learn.adafruit.com/adafruit-neopixel-uberguide/power
Each neopixel needs 20-60mA depending on brightness.  So your 130 LEDs = 2.6-7.8Amps Not something you'll get from a mobile phone charger. So if you need to power them all, you should be looking at something like this: https://www.amazon.co.uk/gp/product/B01DPXDB18

closed 2017-09-16 by jgarff

dushkostanoeski, 2017-10-23:
@shortbloke I finally managed to get my hands on a level shifter, but it is not 74AHCT125, but 74AHCT126. I'm guessing the same wiring won't work because the 1OE pin works on active HIGH and not active LOW. I have no idea how to wire this, so any help would be much appreciated 😄

shortbloke, 2017-10-23:
@dushkostanoeski A quick read suggests that the difference is that with the 125 you have to enable each buffer by pulling it high, where as the 126 they are enabled by default and disabled by pulling them low. "The output pins are enabled by default and can be disabled by pulling down the OE (Output Enable) pins" - Source: https://www.kiwi-electronics.nl/74ahct126-viervoudige-level-shifter-3v-naar-5v?lang=en

dushkostanoeski, 2017-10-23:
I currently have 5V on pin 14, ground on pin 7, input at pin 2 and output at pin 3. Pin 1 is empty. Should I put 5V on that pin as well?

shortbloke, 2017-10-23:
Generally leaving pins floating should be avoided. So connecting pin 1 to Vcc will pull it high and keep it enabled. It's also good practice to pull disable any unused buffers, connecting pins 4, 10 and 14 to Gnd.