jgarff/rpi_ws281x #204: Max led count on RPI3

saoron, 2017-06-29:
Hi guys,
I've connected a NeoPixel ring of 16 LED's to GPIO 10 (SPI) and I only get 8 LED's to work, I've tried to increase spidev.bufsiz but without luck, 

Any ideas?

saoron, 2017-07-19:
@jgarff could you please help out?

Gadgetoid, 2017-08-21:
This may be a question for @tvoverbeek who architected the SPI and PCM support.

tvoverbeek, 2017-08-21:
@saoron 8 LEDs need 96 bytes, which is handled by the spidev driver without DMA. For more than 8 LEDs the spidev driver uses DMA. Do you have anything else which might use the DMA channel used by SPI?
I have tested with an 8x8 array (64 LEDs) and an Xmas string of 240 LEDs. Worked fine on SPI.
I am away from my PIs and cannot check if recent OS updates might have broken the code.
Which version of Raspbian (assuming you are using Raspbian) are you using?

saoron, 2017-08-22:
Hi @tvoverbeek, Thanks for helping out.
I'm using the latest version of Raspbian (lite)

tvoverbeek, 2017-09-03:
Did some testing and it works OK for me with 64 LEDs on SPI, but ....
When using a Pi 3 or Pi Zero W (on-board WiFi and Bluetooth) the variable core frequency can make the SPI timing not work. You will have to fix the core frequency by adding 'core_freq=250' to /boot/config.txt. Then SPI works both on Jessie and Stretch.
Using PCM does not work under Stretch. Do not know why.
PCM under Jessie works.

saoron, 2017-09-14:
Hi @tvoverbeek,
I've ordered a new neopixel and it looks like it works as expected, I guess it was a defected ring,
Sorry for the trouble and thank you for your reply.

closed 2017-09-14 by saoron