jgarff/rpi_ws281x #205: Java wrapper class for this lib

konsultaner, 2017-07-01:
Hey! Thank you for your great work! I currently try to build my own [wordclock](http://www.instructables.com/id/Und-noch-eine-Wordclock/) using the NaoPixels. I wrote a [Java-Wrapper](https://github.com/konsultaner/word-clock-raspberry-pi-zero-neopixels) for your library. Maybe you want to link my project with the hint of an existing Java Wrapper.

jgarff, 2017-08-24:
Awesome!  Thanks.  That's great.  One question however as I'm wondering why you needed to use the gpio_alt.c code.  I believe with the right defines, this should be taken care of by the ws281x library itself.  Maybe I'm missing something.

konsultaner, 2017-08-24:
@jgarff Oh, I didn't know that. I followed the instructions from the adafriut website, where they described how to bring audio to the Pi Zero. I did not go into detail with the ws281x lib at this point.

limpygnome, 2017-09-19:
I've managed to do this using swig:
https://github.com/limpygnome/build-tv/tree/master/libs/neopixel-ws281x-lib

mattjlewis, 2017-10-29:
FYI another Java Wrapper (using JNI): https://github.com/mattjlewis/diozero/tree/master/diozero-ws281x-java

mbelling, 2018-02-26:
I was looking for a java wrapper for this and came across this issue.  I looked at the solutions proposed here and liked the SWIG approach.  That solution however was embedded in a larger project, so I decided to take that concept and create my own standalone library based on it.  I also used some code to load in the generated .so file, as well as a convenience bash script to build that .so file remotely.  The source for my library can be found here: https://github.com/mbelling/rpi-ws281x-java.

I also pubished an initial version to maven central, so build tools like maven and gradle can pull it in: http://search.maven.org/#artifactdetails%7Ccom.github.mbelling%7Crpi-ws281x-java%7C1.0.0%7Cpom

hackerjimbo, 2018-04-19:
> @mbelling <https://github.com/mbelling> since you seem to have the
> most recent and standalone Java wrapper, would you care to bring it
> under the |rpi-ws281x| umbrella to draw more attention and, hopefully,
> ongoing contributions to it?
>
> See: #267 <https://github.com/jgarff/rpi_ws281x/issues/267>
>
> It would live under here: https://github.com/rpi-ws281x

I've also got a Java wrapper for the library. Sounds like we should swap 
notes and consolidate! Check out 
https://github.com/hackerjimbo/PiJava/tree/master/Jimbo/Devices/WS2811

penfold42, 2018-04-19:
@gadgetoid 
https://github.com/rpi-ws281x/rpi-ws281x-java

Gadgetoid, 2018-08-29:
I think all Java discussion should now move to the Java repository, so I'll close this Issue for the time being!

closed 2018-08-29 by Gadgetoid