jgarff/rpi_ws281x #207: Individual neopixels

mitchcoop16, 2017-07-13:
Is there a way to control each neopixel individually while keeping the rest off? I have a 24 pixel ring and I'm using it with a ping sensor so when an object gets closer lights will go one 1 by 1 until the whole ring is light up based on the distance of the object. I was able to set the color of a single/section light by using setPixelColor() but it was unstable. The rest of the lights would just flicker randomly and even the section wouldn't remain the same color. Just looking for some advice!

penfold42, 2017-07-13:
Is pwm audio disabled ?

Do you have a level shifter on the data line ?

mitchcoop16, 2017-07-13:
@penfold42 
I have the audio disabled and I do have a level shifter. 
I was able to do a simple colorwipe() based on the distance with no problems, it just happens when I try to do individual lights.

mitchcoop16, 2017-07-13:
@penfold42 
Okay, I actually fixed the flickering issue and got it all to work.  I think it was just a logic error.

closed 2017-09-16 by jgarff