jgarff/rpi_ws281x #208: sd card becomes read-only after using python3 neopixel library

marcwagner, 2017-07-27:
I'm getting the following error after testing the neopixel library for the first time when using tab-complete:

    -bash: cannot create temp file for here-document: Read-only file system

This happened a few days earlier as well and I thought it was due to a corrupt SD card. However, after a fresh install with a new sd card, I now think it may be the library.

for the record, I am running a RPi3B v2, raspbian-2017-06-23 and python 3.4.2

my guess is that this is an issue with DMA.

two questions I have:
    - is it possible to salvage this install (it's pretty fresh, so no worries if I can't)
    - Is this a bug (I suspect it is), and if so, how do I go about addressing and or fixing it?

here is the /var/syslog data:

`Jul 27 19:23:35 raspberrypi kernel: [ 1593.466401] mmc0: sbc op 23 arg 0x40 flags 0x15 - resp 00000000 00000000 00000000 00000000, err 0
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466411] mmc0: cmd op 25 arg 0x45ab60 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466417] mmc0: data blocks 40 blksz 200 - err 0
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466427] mmc0: stop op 12 arg 0x0 flags 0x49d - resp 00000000 00000000 00000000 00000000, err 0
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466431] mmc0: =========== REGISTER DUMP ===========
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466436] mmc0: SDCMD  0x00000099
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466440] mmc0: SDARG  0x0045ab60
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466445] mmc0: SDTOUT 0x017d7840
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466449] mmc0: SDCDIV 0x00000003
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466454] mmc0: SDRSP0 0x00000900
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466459] mmc0: SDRSP1 0x00001918
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466463] mmc0: SDRSP2 0xffffffff
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466468] mmc0: SDRSP3 0x0002400f
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466472] mmc0: SDHSTS 0x00000001
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466477] mmc0: SDVDD  0x00000001
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466481] mmc0: SDEDM  0x0001080a
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466486] mmc0: SDHCFG 0x0000040e
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466491] mmc0: SDHBCT 0x00000200
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466495] mmc0: SDHBLC 0x00000040
Jul 27 19:23:35 raspberrypi kernel: [ 1593.466499] mmc0: ===========================================
Jul 27 19:23:35 raspberrypi kernel: [ 1593.467831] bcm2835-dma 3f007000.dma: DMA transfer could not be terminated
Jul 27 19:23:35 raspberrypi kernel: [ 1593.473701] mmcblk0: error -110 transferring data, sector 4565856, nr 64, cmd response 0x900, card status 0xc00
Jul 27 19:23:46 raspberrypi kernel: [ 1603.704839] mmc0: timeout waiting for hardware interrupt.`

shortbloke, 2017-07-27:
If you reboot does it work again? I found that switching to use DMA 10 fixed the problem caused by running on the latest raspbian version. Per: https://github.com/jgarff/rpi_ws281x/issues/188

marcwagner, 2017-07-27:
rebooting does not fix the read-only issue. I will try to use dma 10 instead of the default value.

Other than a complete reinstall, how can I fix the read-only issue?

shortbloke, 2017-07-27:
If the card really has gone read only, then I'd suspect the card is failing. You maybe able to try backing the card up with win32diskimager (or simialr), then formatting it with SDFormatter and re-writing the image and see it will take it.  I know you said you tried a new card, but were they part of the same batch, there have been some fake cards which claim to be bigger than they really are which cause problems. Ensure you can write and read back a large file.

Hopefully your issue is just you are hitting the DMA issue quickly if the have the library loading on startup. For me I had normally a few mins before it would fall over.

marcwagner, 2017-07-27:
ok, will look at a different SD card as well and report back.

marcwagner, 2017-07-28:
I installed raspbian 2017-07-05 on a fresh SD card and used DMA 10. it worked. Looking through syslog, I'm pretty sure it was the DMA that gave the error.

closed 2017-07-28 by marcwagner

marcwagner, 2017-08-12:
extra info: disabling the sound card also does the trick

see https://github.com/jgarff/rpi_ws281x/issues/177

make-ing, 2017-09-08:
Hello,

i have the same problem but neither changing the DMA channel nor disabling the sound card works for me. Is there a chance to see which dma channels are free?