jgarff/rpi_ws281x #209: GPIO40 Unselectable

keabarnes, 2017-07-31:
On setting LED_PIN = 40, the following error is received after running strandtest.py:

`Gpio 40 is illegal for LED channel 0
Traceback (most recent call last):
  File "strandtest.py", line 87, in <module>
    strip.begin()
  File "build/bdist.linux-armv7l/egg/neopixel.py", line 115, in begin
RuntimeError: ws2811_init failed with code -11 (Selected GPIO not possible)`

Looking at the code, it would seem the number 40 needs to be added to gpionums_40p array in ws2811.c.

chuchaki, 2017-08-20:
Hey,

I have exactly the same error, only that it already appears using the default pwm (gpio18). Im using an old rpi B rev 2 (2011.12). Are there any known issues that could cause that problem?

chuchaki, 2017-08-21:
Follow up:
The error is thrown by the method '_static int check_hwver_and_gpionum(ws2811_t *ws2811)' int the file 'https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c]'  at line 665.

It separates by the corresponding hardware revision code (/proc/cpuinfo). This list ( http://www.raspberrypi-spy.co.uk/2012/09/checking-your-raspberry-pi-board-version/ ) states that model A has the revision number 0x008. The code in line 687 indicates that the model A is selected by the else if branch. 0x00e - 0x00f indicates, that only model B rev 2 is possible here. Is this an error, or is model A not supported.

tvoverbeek, 2017-09-12:
@chuchaki  See my PR #222. Model A should be recognised now.

tvoverbeek, 2017-09-13:
@keabarnes Also included CM1 and CM3 support in PR #222.
Cannot test it myself, but GPIO40 should work with this PR

jgarff, 2017-09-16:
Merged into mainline.  Please retry with the latest and reopen if necessary.  Thanks.

closed 2017-09-16 by jgarff