jgarff/rpi_ws281x #210: Can't open device file /tmp/mailbox-xxxx: No such device or address in virtual environment

TimonGaebelein, 2017-08-02:
I created a virtual environment so I can debug my code with PyCharm. I set everything up, installed the dependencies and changed the owner of python in the virtual environment to root. However if I start the program from my windows machine I get the following error messages:

```
Can't open device file /tmp/mailbox-1177: No such device or address
Can't open device file /tmp/mailbox-1177: No such device or address
Can't open device file /tmp/mailbox-1177: No such device or address
Traceback (most recent call last):
  File "../Raspberry/main.py", line 24, in <module>
    config.get("led-strip", "pwm-channel"), config.get("led-strip", "type"))
  File "../Raspberry/helpers/led_strip.py", line 30, in __init__
    self.strip.begin()
  File "../virtual/lib/python3.6/site-packages/neopixel.py", line 100, in begin
    raise RuntimeError('ws2811_init failed with code {0}'.format(resp))
RuntimeError: ws2811_init failed with code -1
DMA Error: a8a7c07f
bash: line 1:  1177 Segmentation fault      env "PYTHONUNBUFFERED"="1" "PYTHONPATH"="..:../Raspberry:../Backend:../Documentation/doc:../Images/Image Farming Tool" "PYCHARM_HOSTED"="1" "JETBRAINS_REMOTE_RUN"="1" "PYTHONIOENCODING"="UTF-8" ../virtual/bin/python3.6 -u ../Raspberry/main.py

Process finished with exit code 139
```
The program runs completely fine, if I call it directly via the terminal. Would be great if anyone could help me!

sschroe, 2017-08-15:
The library needs to run on native RPI hardware. Running it in a virtual environment on Windows or such is not going to work. Though i'm not sure if i understood that correctly.

jgarff, 2017-08-24:
Agreed.  I'm going to close this issue.  The RPi hardware isn't present in a virtual environment so this unfortunately won't work.  In this case, it's looking specifically for the Broadcom video device, which is used for DMA buffers.

closed 2017-08-24 by jgarff

charliesneath, 2019-02-06:
> The RPi hardware isn't present in a virtual environment so this unfortunately won't work.

@jgarff Would this also apply when running within a Docker image in an RPi?