jgarff/rpi_ws281x #212: External Soundcard in conjunction with this lib

konsultaner, 2017-08-11:
As you need to disable the internal soundcard, I thought I may be able to use an external device. Is there any experience with the usage of this lib in conjunction with an external [soundcard](https://www.amazon.de/Adapter-Notebook-Ultrabook-Tablet-Windows/dp/B00NOIK9QA)?

siggy, 2017-08-11:
@konsultaner i use this device: https://www.amazon.com/dp/B00NMXY2MO/ref=cm_sw_r_cp_apip_pjHNKTnZPqvRh

example usage: https://github.com/siggy/bbox

konsultaner, 2017-08-11:
@siggy thank you! this is exactly what I was looking for!

closed 2017-08-11 by konsultaner

marbaque, 2017-11-07:
Can use neopixel lib + audio output using this device?

d21d3q, 2018-02-24:
I managed to use external sound card (with pulsaudio), but I had to [disable](http://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/) internal sound sound card, because even with having set external as default, I was having library issue.

I am referencing #50, #47, because they refer to same problem (and solution), #229 was solved in the same way.