jgarff/rpi_ws281x #215: try to render 2 led strips with PWM but one of them keep flickering

agalloch21, 2017-08-18:
I'm using RaspPi 3 modelB, and trying to control 2 led strips using PWM0(18) and PWM1(13). but when I rendered, one of the strips kept flickering, more specifically,

1) when I render 1 or 2 pixels, the led controlled by PWM0 kept flickering and the 3rd pixel also flickered, the led controlled by PWM1 worked fine. 
2) when I render 3 or 4 pixels, PWM0 worked fine, PWM1 kept flickering and the 5th pixel also flickered.
3) ... ....

Sorry don't know if it's my problem but I checked all I can check.

penfold42, 2017-08-18:
Code sample ?

agalloch21, 2017-08-18:
I was using a nodejs module when I post this issue and it' hard to give a code sample. Then I modified  main.c included in the source to make a simple sample, but the situation became a little different, 

1. when I render 1 or 2 pixels, the led controlled by PWM0 flickered sometimes ,the led controlled by PWM1 worked fine.
2. when I render 3 or 4 pixels, PWM0 worked fine, PWM1 kept flickering.

### Here is the code sample:
```
#define TARGET_FREQ             WS2811_TARGET_FREQ
#define GPIO_PIN                18
#define DMA                     5
#define STRIP_TYPE              WS2811_STRIP_GBR		// WS2812/SK6812RGB integrated chip+leds

#define WIDTH                   16
#define HEIGHT                  1

ws2811_t ledstring =
{
    .freq = TARGET_FREQ,
    .dmanum = DMA,
    .channel =
    {
        [0] =
        {
            .gpionum = GPIO_PIN,
            .count = LED_COUNT,
            .invert = 0,
            .brightness = 255,
            .strip_type = STRIP_TYPE,
        },
        [1] =
        {
            .gpionum = 13,
            .count = LED_COUNT,
            .invert = 0,
	    .brightness = 255,
	    .strip_type = STRIP_TYPE,
        },
    },
};

ws2811_led_t rgb2Int(int r, int g, int b) {
	return ((r & 0x00ffffff) << 16) + ((g & 0x00ffffff) << 8) + (b & 0x00ffffff);
}

void led_render(int offset)   // a new function to change brightness 
{
	int x, y;
	int c1 = offset;
	int c2 = (offset+128)%256;
	for (x = 0; x < width; x++)
    {
        for (y = 0; y < height; y++)
        {
			ledstring.channel[0].leds[(y * width) + x] = rgb2Int(c1, c1, c1);
			ledstring.channel[1].leds[(y * width) + x] = rgb2Int(c2, c2, c2);
        }
    }
}

void led_clear()
{
	int x, y;
	for (x = 0; x < width; x++)
    {
        for (y = 0; y < height; y++)
        {
			ledstring.channel[0].leds[(y * width) + x] = 0;
			ledstring.channel[1].leds[(y * width) + x] = 0;
        }
    }
}
```

### loop in main function
```
while (running)
    {
        //matrix_raise();
        //matrix_bottom();
	//matrix_render();
		led_render(offset);
		offset = (offset+1)%256;
		

        if ((ret = ws2811_render(&ledstring)) != WS2811_SUCCESS)
        {
            fprintf(stderr, "ws2811_render failed: %s\n", ws2811_get_return_t_str(ret));
            break;
        }

        // 15 frames /sec
        usleep(1000000 / 15);
    }
```

qwertyflagstop, 2018-09-22:
Did you ever figure out why one strip was flickering?

Gadgetoid, 2022-01-25:
Closing, since this issue is stale even by the most charitable of estimations.

:robot:

closed 2022-01-25 by Gadgetoid