jgarff/rpi_ws281x #217: Debian Package for rpi_ws281x

Gadgetoid, 2017-08-21:
Since Pimoroni is possibly the largest user of the rpi_ws281x library, with our Unicorn HAT and Unicorn pHAT products, I'm very happy to be the one who maintains and deploys a debian package for `rpi_ws281x`. This would be uploaded to the Raspberry Pi repository along with our currently shipped Python packages.

My reasoning? Well, I want to ship `rpi_ws281x` - the Python library - as a Debian package, but I find it unconscionable that I should do it with a baked-in, statically-linked version of `rpi_ws281x` - it would be rude, to say the least. So, I'd like to ship a library package (which all users could potentially benefit from), that I can then link the Python library against.

To do this, I'll need a `debian` folder in this repository containing all the relevant information for packaging and distributing it. I would then be happy to validate, package, sign and distribute the result.

I'm currently in the process of figuring out how to do this myself, but any help would be appreciated.

jgarff, 2017-08-24:
Thanks @Gadgetoid!  I think this is a great idea!  It's something I've had on my list as well.  I'll try to get some time this weekend to help work on this.  Creating the debian folder along with the associated metadata should be very straight forward.

Gadgetoid, 2017-08-25:
Great news. I had made a very small start on the debian folder and metadata, but with little prior experience it's something that requires more patience and precision than I could muster at the moment. I'm glad you're on board!

jgarff, 2017-08-27:
I just checked in bd32bedf446a609528da9a797a1a7e4852decd5a to build a shared library Raspbian package.  Please give it a shot.  We should probably also add a dev package as well that includes the required headers, but please give this a shot and let me know what you think.

To build the package,  you will need to use the "scons deb" command.

Gadgetoid, 2017-08-27:
Builds and installs beautifully, good stuff!:

```
pi@raspberrypi:~/Development/jgarff/rpi_ws281x $ scons deb
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...

cd /home/pi/Development/jgarff/rpi_ws281x; dpkg-deb --build libws2811_1.1.0-1
dpkg-deb: building package 'libws2811' in 'libws2811_1.1.0-1.deb'.
scons: done building targets.

pi@raspberrypi:~/Development/jgarff/rpi_ws281x $ sudo dpkg -i libws2811_1.1.0-1.deb
Selecting previously unselected package libws2811.
(Reading database ... 159610 files and directories currently installed.)
Preparing to unpack libws2811_1.1.0-1.deb ...
Unpacking libws2811 (1.1.0-1) ...
Setting up libws2811 (1.1.0-1) ...

pi@raspberrypi:~/Development/jgarff/rpi_ws281x $ ls /usr/lib/libws*
/usr/lib/libws2811.so
```

A dev package would, indeed, be required to make any use out the library. I ran into some niggles with trying to use just `ws2811.h`, however, since it depends upon `pwm.h` and `rpihw.h`.

jgarff, 2017-08-28:
Yeah, I think we need to clean up the header files before they can be installed globally on a system.  That's one of the reasons I didn't keep going and do a dev package as well, the other reason being time.  I also didn't get finished with reviewing your pull request, so please stay tuned on that.  Thanks again for all your help!

Gadgetoid, 2017-08-29:
Stuff like this takes time! I appreciate that. Thanks for your efforts thus far- not to mention this library in the first place!

I've now shifted from my hard-fork of rpi_ws281x (used because I needed to slip in some extra features and stay ahead of new Pi releases) to a regular fork, and my own Python bindings for that live here: https://github.com/pimoroni/rpi_ws281x-python

I've actually deprecated the "neopixel" naming scheme from this version of the Python library (although keep a compatibility shim for it) in favour of "rpi_ws281x"

Hopefully I'll be able to replace the slightly contrived submodule here with a link against the packaged library.

Since I was publishing this library to https://pypi.python.org/pypi/rpi_ws281x I was keen to get rid of my kludgy hard fork and get it as close to the original upstream code as possible- since it runs the risk of causing invalid bug reports and all manner of confusion (I found this out the hard way with WiringPi)

Frodox, 2017-11-28:
@Gadgetoid  So, can I install it just by `pip3 install rpi_ws281x` now ?

Frodox, 2017-11-28:
Well, it works, but how to use it then :D

Gadgetoid, 2017-11-29:
The `rpi_ws281x` pip package has been available for some years now, however it's still built in a detached state from this repository and maintained at my convenience primarily for supporting our Unicorn HAT and Unicorn pHAT products.

I have made significant efforts to merge our customisation upstream and bring the release closer to what you find here, but:

* It still lags behind the latest changes here
* There's no release schedule for this library, against which to mark rpi_ws281x PyPi releases

So, if you have any issues with it, please raise them at https://github.com/pimoroni/rpi_ws281x-python

Or against our fork or rpi_ws281x https://github.com/pimoroni/rpi_ws281x

Gadgetoid, 2021-07-13:
It's now possible to build a deb package using CPack by adding something like the following to CMakeLists.txt:

```cmake
set(CPACK_PACKAGE_CONTACT "Jeremy Garff (jer@jers.net)")
include(CPack)
```
And running:

```
mkdir build
cd build
cmake ..
sudo cpack -G DEB  # sudo needed due to the build process running ldconfig, might want tweaking
```