jgarff/rpi_ws281x #218: Error initializing the strip

Margravegit, 2017-08-23:
Hi!
While trying to run any of the examples, I'm encountering this error:

strip.begin()
file "build/bdist.linux-arm7l/egg/neopixel.py, line 114, in begin
attributeerror: ´module´ object has no attribute 'WS2811_SUCCESS'

Followed the steps on the readme with lo luck. any ideas?

Margravegit, 2017-08-23:
Using a raspberry pi zero w and a fresh noobs install

sschroe, 2017-09-01:
Please provide more information on the commands you are using and which OS you use etc.

gitBugs, 2017-09-16:
I had this same issue, which I've now fixed. On a Pi Zero W running a clean install of Raspbian GNU/Linux 8 (jessie) I followed the instructions in the two readme files:
First, the build instructions in the parent folder:
```
~/Downloads/rpi_ws281x $ apt-get install scons
~/Downloads/rpi_ws281x $ scons
```
..then confirmed that the test worked
`~/Downloads/rpi_ws281x $ sudo ./test`
...then built the python wrapper:
```
  ~/Downloads/rpi_ws281x $ sudo apt-get install python-dev swig
  ~/Downloads/rpi_ws281x $ python ./setup.py build
```
...and tried to run a Demo:
` ~/Downloads/rpi_ws281x $ sudo PYTHONPATH=".:build/lib.linux-armv7l-2.7" python examples/strandtest.py`

...which yielded the same error that @Margravegit reported above.

I fixed it by running 
` ~/Downloads/rpi_ws281x $ sudo python ./setup.py install `
...which I saw mentioned in another thread on this issues page. Now the command to run a demo works as expected.

If running `setup.py install` is necessary for using the python wrapper, perhaps it should be added to the readme?

Gadgetoid, 2018-03-06:
Looks like these instructions have been added as of 81af569adbfd7978caeee12f06eb9863d1e57dc6

closed 2018-03-06 by Gadgetoid