jgarff/rpi_ws281x #223: Library works on Pi3 but nog Pi Zero (1.3)

prutsky, 2017-09-17:
I wrote a python program on a Raspberry Pi 3 that worked fine. 
Now I want this code to run on a Raspberry Pi Zero (1.3), and it won't rund fine. The 8 LEDs on the stick won't turn on. When I kill the program (CTRL+C) all LEDs turn on bright white.
I tested this with the same SD card on multiple Pi Zero's (v1.3).
On a Pi Zero W a bit more happened but still not the same as it should (as on the rPi3). On the Zero W the LEDs turn on, but nog much color.

Then I checked an older Jessie image, and there the LED do work fine. The differences are (uname -a):
Not working Pi Zero (v1.3) with Jessie:
Linux Cat-O-Matic 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux

Working Pi Zero (v1.3) with Jessie:
Linux Kattenluikje 4.4.50+ #970 Mon Feb 20 19:12:50 GMT 2017 armv6l GNU/Linux

Anybody has an idea if this is fixable so it works with the latest Jessie?

penfold42, 2017-09-17:
Try dma channel 10 instead of the default of 5

prutsky, 2017-09-17:
That worked! Thanks.

closed 2017-09-17 by prutsky