jgarff/rpi_ws281x #225: Cannot run strandtest from Python3

ColdMedina, 2017-09-19:
Im a complete newb
Pi3, Raspbian, 12 neopixel ring, audio disabled in config.txt

strandtest works when running as sudo from terminal
strandtest does not work when running from Python3 text editor (pressing F5)

RuntimeError: ws2811_init failed with code -5 (mmap() failed)

It did work yesterday. Not sure what I did, have reloaded fresh raspbian and followed everything exactly...three times now to no avail.

penfold42, 2017-09-19:
Are you running as root ?

ColdMedina, 2017-09-20:
When I run it in terminal, I use sudo and works fine

When in Python3, I press F5 to save and run --- get the error

penfold42, 2017-09-20:
Unless your python3 text editor is running as root, I don't think it will work

ColdMedina, 2017-09-21:
Newb question -- is that possible by clicking from the desktop, or so I need to open via terminal? (and if so, how?)

Frodox, 2017-11-28:
@ColdMedina hi! How did you install/run it as python3?
I have problem after
```
python3 ./setup.py build
```
there is file `build/lib.linux-armv6l-3.5/_rpi_ws281x.cpython-35m-arm-linux-gnueabihf.so`, but 
```
$ sudo PYTHONPATH=".:build/lib.linux-armv6l-3.5" python examples/strandtest.py -c
    from neopixel import *
  File "/home/pi/workspace/rpi_ws281x/python/neopixel.py", line 5, in <module>
    import _rpi_ws281x as ws
ImportError: No module named _rpi_ws281x
```

thelasthope4U, 2017-12-20:
Does anyone have insight to this issue?  I believe it is a Python3 compatibility issue.

Frodox, 2017-12-20:
@thelasthope4U  Yes, I got it running from python3 and sudo :D

Gadgetoid, 2018-03-06:
You can run your Python 3 editor (this is just what Raspberry Pi call IDLE3 in the drop down menu) as root by opening a Terminal window and typing:

```
sudo idle3
```

You should then be able to hit F5 and run your code without error.

closed 2018-04-17 by Gadgetoid