jgarff/rpi_ws281x #226: Wrong control of 200 RGB WS2811 led string

Max22b, 2017-09-22:
Hi to everyone, I apologize for my English. I have a raspberry 2 with four strings ws2811 5V connected in series for a total of 200 led. I downloaded the library and modified the parameters of the file strandest.py . The power supply is brought to the first and third string. I connect the string at GPIO18 PWM via SN 74AHCT125 and i put a capacitor before power supply.
I cannot control the last 50 LEDs that always appear red weak. Can it be a frequency or power problem?
Thanks to everyone!!!
Massimiliano from Italy :-)

penfold42, 2017-09-22:
You probably are seeing voltage drop along the strip.

I needed to supply power at both ends of my 144 led strip

Max22b, 2017-09-25:
OK thanks. I thought of that problem and I put power supply at the third string but I did not solve it. now I try to feed it in the end and see if I fix it. 
Is it possible a signal data problem? poor data voltage? Thanks a lot

penfold42, 2017-09-25:
If you're wiring is good then the data signal will be fine. Each led regenerates the data signal.

I just reread your original post - does the last strip change and display data but with blue and green missing ?
Or is it a static dim red ?

Max22b, 2017-09-25:
no no, the last 50 led don't display the data...but a pooooor  red light,,,  
I also changed string but gets the same problem. thanks

Max22b, 2018-01-03:
Resolved. it was a power supply problem and string with bug. thank you!!!!

closed 2018-01-03 by Max22b