jgarff/rpi_ws281x #227: Most Basic setup question

PreciousRoy0, 2017-09-23:
Hi all,

I have been having some trouble connecting a led-ring RGBW or a RGB ring to my Pi 3,
I downloaded the code changed the amount of leds and built it,  and start the program as mentioned : sudo ./test -s rgbw

i connected the ledrings to 5v gnd and directly to pin 18 (pin 12), although not recomended it should work for 16 leds right.

but all i get is a semi reandom led flashing with ether led rings. i say semi random becouse there is a patern to the flashing its always the same.

i did not disable the audio because the file mentioned does not exist on my pi but i am not using it ether. i am running rasbian 1.3.

do any of you have an idea what i am doing wrong?

sschroe, 2017-09-27:
Disable the audio and if that still doesn't help add a level shifter or power the led ring with 3,3V only.

PreciousRoy0, 2017-10-01:
thank you,

i am still having the same problem when powering the ledring with 3v. 
so i tried turning off the audio device....

`Since this library and the onboard Raspberry Pi audio both use the PWM, they cannot be used together. You will need to blacklist the Broadcom audio kernel module by creating a file /etc/modprobe.d/snd-blacklist.conf with

blacklist snd_bcm2835
If the audio device is still loading after blacklisting, you may also need to comment it out in the /etc/modules file.

On headless systems you may also need to force audio through hdmi Edit config.txt and add:

hdmi_force_hotplug=1
hdmi_force_edid_audio=1
A reboot is required for this change to take effect

Some distributions use audio by default, even if nothing is being played. If audio is needed, you can use a USB audio device instead.`

but the files in this explanation do not exist. and i am a linux noob and google gives 1000 ways of doing it but i have no clue it its the same. and i will try the level sifter.

PreciousRoy0, 2017-10-01:
Ok after a lot of testing and reading this forum post making the code a lot more readable i edited it to this. 

```
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <sys/mman.h>
#include <signal.h>
#include <stdarg.h>
#include <getopt.h>

#include "clk.h"
#include "gpio.h"
#include "dma.h"
#include "pwm.h"
#include "version.h"

#include "ws2811.h"

#define TARGET_FREQ                             WS2811_TARGET_FREQ
#define GPIO_PIN                                18
#define DMA                                     5
//#define STRIP_TYPE                              WS2811_STRIP_RGB        // WS2812/SK6812RGB integrated chip+leds
//#define STRIP_TYPE                            WS2811_STRIP_GBR        // WS2812/SK6812RGB integrated chip+leds
//#define STRIP_TYPE                            SK6812_STRIP_RGBW       // SK6812RGBW (NOT SK6812RGB)

//#define STRIP_TYPE  WS2811_STRIP_GRB //<------ neopixel setting
#define STRIP_TYPE  SK6812_STRIP_GRBW  //<----- neopixel rgbw setting

#define LED_COUNT                               8

ws2811_led_t dotcolors[] =
{
    0x00200000,  // red
    0x00201000,  // orange
    0x00202000,  // yellow
    0x00002000,  // green
    0x00002020,  // lightblue
    0x00000020,  // blue
    0x00100010,  // purple
    0x00200010,  // pink
	};
	
ws2811_led_t dotcolors_rgbw[] =
{
    0x00200000,  // red
    0x10200000,  // red + W
	
	0x00002000,  // green
    0x10002000,  // green + W

    0x00000020,  // blue
    0x10000020,  // blue + W
	
    0x00101010,  // white
    0x10101010,  // white + W

};

ws2811_t ledstring =
{
    .freq = TARGET_FREQ,
    .dmanum = DMA,
    .channel =
    {
        [0] =
        {
            .gpionum = GPIO_PIN,
            .count = LED_COUNT,
            .invert = 0,
            .brightness = 255,
            .strip_type = STRIP_TYPE,
        },
    },
};

int main()
{
    ws2811_init(&ledstring);

    int c = 0;
	int i = 0;
	
	int running = 1;
	
	while(running)
	{   
		for (i = 0; i < LED_COUNT; i++)
		{
			if (ledstring.channel[0].strip_type == STRIP_TYPE) {			
				if(c == i)
					ledstring.channel[0].leds[i] = dotcolors_rgbw[c];
				else
					ledstring.channel[0].leds[i] = 0x00000000;
			}
			else
			{
				if(c == i)
					ledstring.channel[0].leds[i] = dotcolors[c];
				else
					ledstring.channel[0].leds[i] = 0x00000000;
			}
			ws2811_render(&ledstring);
		}
		c++;
		if(c > LED_COUNT)
			c = 0;
		usleep(1000000);
	}
    return 0;
}
```
first problem, 
//#define STRIP_TYPE  WS2811_STRIP_GRB //<------ neopixel setting
#define STRIP_TYPE  SK6812_STRIP_GRBW  //<----- neopixel rgbw setting

my corrects settings i did not know.
as for your test code even set to the correct strip type it still stays all white.

Gadgetoid, 2021-12-03:
Closing since this is stale and I like the number of issues to go down sometimes :laughing: instead of just up :sob: 

Feel free to re-open if it's still relevant.

closed 2021-12-03 by Gadgetoid