jgarff/rpi_ws281x #228: Raspberry pi 3 Neopixel Ring SPI (gpio 10): Segmentation fault

thread9000, 2017-09-23:
(Running on Stretch)
```
# LED strip configuration:
LED_COUNT      = 12      # Number of LED pixels.
LED_PIN        = 10      # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ    = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA        = 5       # DMA channel to use for generating signal (try 5)
LED_BRIGHTNESS = 255     # Set to 0 for darkest and 255 for brightest
LED_INVERT     = False   # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL    = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53
LED_STRIP      = ws.WS2811_STRIP_RGB   # Strip type and colour ordering
```

When I run `sudo python strandtest.py` it returns: 
```
Press Ctrl-C to quit.
Color wipe animations.
Segmentation fault
```

I can see the /dev/spidev0.0 and the ring is running off of 5v.

Any suggestions on debugging this?

forevertheuni, 2017-10-03:
I have the same problem. 

`./test 
Can't open /dev/mem: Permission denied
ws2811_init failed: Unable to initialize SPI
pi@hera:~/rpi_ws281x $ sudo ./test 
Segmentation fault
pi@hera:~/rpi_ws281x $ ls /dev/spidev0.
spidev0.0  spidev0.1

JIP0815, 2017-10-07:
Same here with my fresh installed Pi Zero.
Latest version of raspbian (stretch lite) and firmware.
Using Python 2.7.13.
Default PIN 18 (PWM) works fine.

jgarff, 2017-10-08:
Fix merged.  Please re-test with the latest master branch and re-open if this is still a problem.  Thanks.

closed 2017-10-08 by jgarff

apryjom, 2017-10-17:
I can still see 'Segmentation fault' when stop e.g. lowlevel.py on my pi zero.

apryjom@pizero:~/rpi_ws281x-master/python/examples $ sudo python lowlevel.py
^CTraceback (most recent call last):
File "lowlevel.py", line 87, in
time.sleep(0.25)
KeyboardInterrupt
Segmentation fault