jgarff/rpi_ws281x #230: Hugues voltage loss and RGBW issue

resetscoder, 2017-10-05:
Hi ! I've just bought an Sk6812 rgbw  in order to build a sunrise alarm clock with a Raspberry Pi B. The led strip is 5m with 60led/m, so that I have a 20A/5V power supply (60mA per led). But I'm facing with some issues while using the white led ! More than half of the strip suffer from à Red shift and intensity of white is lowering to zero when I try to go from color(0,0,0,0) to color (255,255,255,255). Even if I plug the power from the two side of the strip it suffer the same in the center.

My wiring is done with breadboard, May it have an impact on the loss of voltage ?

Is someone facing the same issue ?
It seems to work better with SK6812_lowlevel.py (no lowering of Rgb (without white))

penfold42, 2017-10-05:
Breadboard wire is terrible for power connections.

Both the power and ground wires need to be good size cables.
Mains wiring has to carry 10+ amps - look how much thicker the conductors are.

resetscoder, 2017-10-05:
Thanks for the tips ! I will make the setup with some good wires !

closed 2018-03-06 by Gadgetoid